rancilio-pid/clevercoffee #19: Fehler in moving average 1. readIndex++ in else-Zweig gelegt 2. if geaendert von numReadings - 1 auf 0

gerd-li, 2019-10-09:
`void movAvg() {
  if (firstreading == 1) {
    for (int thisReading = 0; thisReading < numReadings; thisReading++) {
      readingstemp[thisReading] = Input;
      readingstime[thisReading] = 0;
      readingchangerate[thisReading] = 0;
    }
    firstreading = 0 ;
  }
  
  readingstime[readIndex] = millis() ;
  readingstemp[readIndex] = Input ;

  if (readIndex == 0) {    // geaendert numReadings - 1 auf 0
    changerate = (readingstemp[0] - readingstemp[numReadings - 1]) / (readingstime[0] - readingstime[numReadings - 1]) * 10000;
  } else {
    changerate = (readingstemp[readIndex] - readingstemp[readIndex - 1]) / (readingstime[readIndex] - readingstime[readIndex - 1]) * 10000;
  }

  readingchangerate[readIndex] = changerate ;
  total = 0 ;
  for (i = 0; i < numReadings; i++)
  {
    total += readingchangerate[i];
  }

  heatrateaverage = total / numReadings * 100 ;
  if (heatrateaveragemin > abs(heatrateaverage)) {
    heatrateaveragemin = abs(heatrateaverage) ;
  }

  if (readIndex >= numReadings - 1) {
    // ...wrap around to the beginning:
    readIndex = 0;
  }
  else{
  	readIndex++;
  }
}
`

hellgelino, 2020-02-09:
Hi,

ist das schon gefixed?
Und das OTA ist ja eher halb auskommentiert, muß das noch was gemacht werden?

gerd-li, 2020-02-10:
Yes, it‘s allready fixed. My code can be incorporated in the project.

murmeltier08, 2020-12-28:
Should be solved

closed 2020-12-28 by murmeltier08