rancilio-pid/clevercoffee #38: ShotTimer

btotherunner, 2020-12-31:
Wäre es möglich einen ShoTimer zu integrieren?

Damit auf dem Display auch bei der Brüherkennung der Shottimer startet und die Sekunden "Groß" angezeigt werden.

murmeltier08, 2021-01-10:
Wollen wir das per Ini aktivierbar machen?

btotherunner, 2021-01-11:
ja wäre super!

murmeltier08, 2021-01-11:
Dann bau ich mal sowas ein bei Gelegenheit

murmeltier08, 2021-01-11:
Idee: gleichzeitig BD nur noch +/- 1 am Sollpunkt aktivieren.

ben-coe, 2021-01-16:
Könnte man hier direkt auch die Anzeige des Shottimers einige Sekunden stehen lassen (bei Hardware brewdetection), damit man sieht wie lange der Bezug gedauert hat?

murmeltier08, 2021-01-21:
Erste Version ist umgesetzt!

rasimriver, 2021-01-24:
Ich wollte den Shottimer testen, habe ihn aber wieder deaktivert. Jetzt scheine ich ein Bug zu haben: Der Shottimer flackert nun ständig neu auf. Muss man hier beim Hochladen etwas beachten?

murmeltier08, 2021-01-24:
Das sollte eigentlich nicht sein. Könntest du deine Display Config dazu zeigen?

rasimriver, 2021-01-24:
Ja, ich habe den aktuellsten commit verwendet. Die Userconfig habe ich auf Mattermost gepostet [Link](https://chat.rancilio-pid.de/ranciliopid/pl/5rk6zn43rinnxm8cpoh5r9hihr)

murmeltier08, 2021-01-24:
Ich habe den Fehler gefunden: der Shottimer war vorher auf Ebene vom "oder", jetzt sollte es gehen

```
         (
            (bezugsZeit > 0 && ONLYPID == 1) || // Bezugszeit bei Only PID  
            (ONLYPID == 0 && brewcounter > 10 && brewcounter <= 42) // oder Bezug bei nicht only PID über brewcounter
            ) && SHOTTIMER == 1
```

murmeltier08, 2021-01-24:
Problem hoffentlich gelöst im aktuellen Push, bei mir geht es. Sonst melde dich bitte @rasimriver

closed 2021-01-24 by murmeltier08