rancilio-pid/clevercoffee #56: Licht wenn Temperatur erreicht ist +-5 °C

hansgans007, 2021-01-31:
Einen Ausgang schalten wenn die Temperatur erreicht ist :)

murmeltier08, 2021-01-31:
Welchen Ausgang genau?

hansgans007, 2021-01-31:
ich hab derweil auf 16 geklemmt ETRIGGER oder wie der heißt brauch ich eh nicht

murmeltier08, 2021-02-01:
Reicht dir ein HIGH oder noch was? 
HIGH kleiner 5 Grad
LOW größer 5 Grad

hansgans007, 2021-02-05:
Ja das passt im mittel einfach +-5 grad wäre klasse

tml89, 2022-01-11:
Ist diese funktion bereits implementiert? Plane gerade einen Umbau ohne Display und möchte auch nicht den PERFECT COFFEE PID Fork nutzen.

murmeltier08, 2022-01-11:
Melde dich gerne mal im chat bei uns. das sollte sich schnell einbauen lassen. Dann besprechen wir das gerne im Detail

murmeltier08, 2022-01-11:
Ist in der Alpha 3.0.0 nun drin 
PINLED und LEDON muss in der konfig gesetzt werden

closed 2022-01-11 by murmeltier08