rancilio-pid/clevercoffee #113: Feature: Tastschalter für Kaffeebezug

krisraich, 2021-05-01:
Taster Modus für Kaffeebezug. Beim Vollumbau der Rancilio würde ich gerne den normalen Ein/Aus-Schalter für den Bezug durch einen Taster ersetzen.

Taster kurz Ein: (Preinfusion &) Kaffeebezug
Taster lang ( > 1sek) Ein: Ventil auf & Pumpe ein bis Taster wieder Aus.

lg

R-Matthias, 2021-05-12:
Aber reichen euch nicht die 2 Sekunden zum spülen? Mich würde als zweite Funktion auch der Backflush Prozess interessieren.

murmeltier08, 2021-05-12:
Vorschlag wäre, dass wir es konfigurbar machen. Ich müsste mal wissen wie der Taster im Vollausbau überhaupt aktuell funktioniert - kann man damit den Bezug auch stoppen?

R-Matthias, 2021-05-12:
Also bei mir ist es im Vollausbau so, dass zwei Leitungen von der MCU an den Brühschalter gehn und die 230V dort entlang laufen. Im aktuelles ist es so gemacht, dass nur noch zwei Leitungen von der MCU dort hin gehn.

R-Matthias, 2021-05-12:
Bezug kann auch einfach wieder gestoppt werden. Konfigurierbar hört sich gut an 👍

murmeltier08, 2021-05-12:
Hast du ein Taster zur Hand um es bei dir zu testen, wollen wir im chat weiter dazu schreiben?

murmeltier08, 2021-05-16:
bitte meldet euch, wenn jemand sowas testen könnte, dann passe ich dazu den code an

R-Matthias, 2021-05-16:
hatte die letzten Tage einiges zu tun, ich könnte das schon testen, müsste dann aber wohl den 230V Kreis an meinem Brühschalter noch umgehen und aus dem Schalter einen Taster machen (die passende Feder hab ich vom Einschalttaster übrig, da ich dem zum Schalter in meiner Silvia E umgebaut habe)

R-Matthias, 2021-05-16:
Taster Umbau würde bei mir jetzt funktionieren, habe es gerade versuchsweiße aufgebaut und getestet. Bräucht jetzt nur den passenden Code, damit die MCU nur auf das kurze Signal reagiert und dann meinen Brühvorgang (Präinfusion-Pause-Brühvorgang) mit den definierten Werten abspielt. Die zweite Funktion, wenn der Taster dann länger als 1-2Sec gedrückt wird (eventuell auch konfigurierbar?) wird dann konfigurierbar sein mit einfach nur Bezug bis ein weiteres Signal kommt oder eben dann der Rückspülung, welche ich bevorzugen würde.

lebuni, 2021-05-16:
Wo hast du den Taster denn montiert? An die gleiche Stelle, wo woher der Bezugsschalter war? Schick mal ein Foto!

R-Matthias, 2021-05-16:
Hab es schon wieder zurückgebaut, sonst kann ich morgen in der Früh kein Cappuccino trinken. Aber von Prinzip her hab ich die beiden 230V Leitungen einfach nur zusammengeschlossen und die beiden Leitungen zur MCU auf einen Taster gelegt. Danach lief der Bezug genau so lange wie der Taster gedrückt war.

murmeltier08, 2021-05-16:
https://github.com/rancilio-pid/ranciliopid/tree/brewtrigger
Branch für den Taster

krisraich, 2021-06-22:
@murmeltier08 Eine backflush Routine ist prinzipiell in der FW bereits enthalten. Du muss nur im Blynk einen Schalter mit dem Wert "V40" hinzufügen, um den Backflush Modus zu aktivieren. Siehe: https://github.com/rancilio-pid/ranciliopid/blob/master/rancilio-pid/rancilio-pid/rancilio-pid.ino#L465

@lebuni richtig und wenn ich nach dem Bezug das Sieb kurz ausspülen möchte.

@R-Matthias Ich verstehe nicht ganz was die 230V damit zu tun haben? Der Schalter wird ja im Vollausbau mit der 3.3V vom uC betrieben?

Ich denke ich habe mich zu genau ausgedrückt. Daher habe ich mit meinen unglaublichen Excel Skills ein Impulsdiagramm gezeichnet:

[image]

Die Preinfusion läuft normalerweise 2s mit 5s Pause und dann startet der normale Bezug (normal 25s).
Sprich der ganze Bezug dauert 25s ohne oder 32s mit Preinfusion.
Daher würde ich folgendes vorschlagen:

Wird der Taster kurz angetippt (also unter 2s) dann läuft das normale Bezug-Programm (7s Preinfusion + 25s Bezug).
Wird der Taster gehalten (länger als 2s) dann wird die Preinfusion ausgelassen und die Pumpe + Ventil werden betätigt bis der Taster losgelassen wird.

lg

R-Matthias, 2021-06-22:
Auch im Vollausbau können auf dem Bezugschalter 230V liegen, desswegen wird ja die Lampe entsprechend auch ausgebaut. Die 230 V muss man definitiv ändern, wenn man es mit einem Taster machen möchte.

krisraich, 2021-06-22:
Ja stimmt. Aber so wie ich das am Schaltplan lese, könnte man die 230V auch einfach brücken. Dann liegen am Relais die 230V eben dauerhaft an. Sollte ja egal sein.

Das mit der Lampe ist mir noch gar nicht aufgefallen. Müsste die Lampe nicht zwischen Pin 1 & 5 oder  2 & 4 sein?

R-Matthias, 2021-06-23:
Die Lampe bringt dir nicht viel beim Taster, der is ja nicht mehr wirklich lange gedrückt...

krisraich, 2021-06-28:
Was mir noch eingefallen ist: Es wäre ganz sinnvoll, wenn der Bezug im Tastermodus durch eine erneute kurze Betätigung wieder gestoppt werden kann.

[image]

So macht das zumindest auch DeLonghi.

Ich kann sonst auch ein State-Diagramm Zeichnen, falls das beim Entwickeln hilft.

lg

R-Matthias, 2021-06-28:
Funktioniert aktuell schon mit dem stoppen.

murmeltier08, 2021-07-31:
#131 hier wurde nun der Button weiterentwickelt

krisraich, 2021-12-24:
Hallo

Bin gerade dabei dieses Feature einzubauen. Den Bezug starten per Tastendruck funktioniert, wenn ich den Taster länger als 1sek betätige, passiert nichts. Schaut aus, also ob das Programm in einer Schleife hängt (log im Anhang)

[image]

```
// PID & Hardware
#define ONLYPID 0                 // 1 = Only PID, 0 = PID and preinfusion
#define ONLYPIDSCALE 0             // 0 = off , 1= OnlyPID with Scale
#define BREWMODE 1                 // 1 = NORMAL preinfusion ; 2 = Scale with weight
#define BREWDETECTION 2            // 0 = off, 1 = Software (Onlypid 1), 2 = Hardware (Onlypid 0), 3 = Sensor/Hardware for Only PID 
#define BREWSWITCHTYPE 2           //  1 = normal Switch, 2 = Trigger Switch
#define COLDSTART_PID 2            // 1 = default coldstart values, 2 = custom values via blynk (expert mode activated) 
#define TRIGGERTYPE HIGH           // LOW = low trigger, HIGH = high trigger relay
#define VOLTAGESENSORTYPE HIGH     // BREWDETECTION 3 configuration
#define PINMODEVOLTAGESENSOR INPUT // Mode INPUT_PULLUP, INPUT or INPUT_PULLDOWN_16 (Only Pin 16)
#define PRESSURESENSOR 0           // 1 = pressure sensor connected to A0; PINBREWSWITCH must be set to the connected input!
```

Verwende die Version 2.9.4

lg

murmeltier08, 2021-12-24:
Ich glaube wir haben das längere drücken vom Taster nicht mit einer Funktion belegt. Ich glaube mich zu erinnern, dass unklar war was dann genau passieren soll. lässt sicher noch einbauen.

krisraich, 2021-12-24:
Ja die rede war ja das als Backflush zu verwenden. Ich habe das in einer Rancilio L6 (zweikreiser) eingebaut. Bei Zweikreiser wird das Wasser im Boiler mit einem Durchlauferhitzer aufgeheizt. Daher muss man vor dem Kaffeebezug das Wasser das sich im Durchlauferhitzer befinden ablassen damit die Temperatur für den eigentlichen Bezug passt. 

Sprich: Kaffeemaschine idelt, Dann betätigt man den Taster für 10 Sekunden und lässt das zu heiße Wasser ab (perfekt zum Tassen vorwärmen). Danach tippt man den Taster kurz an um einen normalen Bezug zu starten.

Habe bereits erfolglos versucht den Quellcode anzupassen.

murmeltier08, 2021-12-29:
Lässt du das Wasser über die Brühgruppe ab? Ich frage daher, wie / ob das Ventil geschaltet werden muss.

krisraich, 2021-12-31:
Ja das Wasser wird über die Brühgruppe abgelassen.

murmeltier08, 2021-12-31:
Eigentlich ein manueller Brühvorgang ohne Pause, Preinfusion usw., dann werde ich das mal bei gelegenheit einbauen.

krisraich, 2021-12-31:
Genau. Danke, das wäre super.

murmeltier08, 2022-01-07:
@krisraich: Würdest du eine Codeanpassung testen wollen?