rancilio-pid/clevercoffee #184: Counter for etrigger

github-apo, 2022-01-21:
ist es möglich für den etrigger noch eine Art max_counter einzufügen? Dieser Count definiert die Anzahl der Impulse, die der etrigger ausgelöst wird. Bsp. etriggertime = 1200 und etrigger_max_counter = 8 => die Maschine bleibt 3 Stunden an. Erste Auslösung nach 20 Minuten, dann noch 7 mal => 3 Stunden. Wenn die Maschine kontinuierlich anbleiben soll kann man den counter ja auf "0" oder "100" oder so setzen.

murmeltier08, 2022-01-25:
Wir müssen momentan die Alpha 3.0.0 produktiv bekommen, dann machen wir das gerne! :) Vielleicht ergibt sich nebenbei eine Lücke für die kleine Änderung

fiendie, 2023-02-09:
E-trigger support has been removed from version 4.x and will be replaced by a more comprehensive standby mode in the medium term.

LoQue90, 2023-12-05:
there will be no feature updates for esp8266.
with esp32 there is a internal standby mode, where the cpu of the silvia e has to be removed from the mashine.

closed 2023-12-05 by LoQue90