rancilio-pid/clevercoffee #186: Temp + / - via Hardware Buttons

cottec, 2022-01-24:
To allow easy adjustment of temperature for switching between different coffee roasts without a smartphone it would be handy to have hardware controlled buttons on free pins using the ESP32 built in pull up/down

pinMode(pin, INPUT_PULLUP); 
pinMode(pin, INPUT_PULLDOWN);

pins could be 5&17 (next to each other on the board an unused for pidonly)
if routed outside the machine, combined with a display, it's ground connection might be ideal for using pull-down as the positive input could be +5V which wouldn't be healthy for the board

endym, 2022-01-26:
In fact I've implemented such feature already. This [branch](https://github.com/endym/ranciliopid/tree/my_master) is running fine in my Rancilio Silvia E with a NodeMCU. [Here](https://chat.rancilio-pid.de/ranciliopid/pl/49pb6s3g53grumd6mnn16g9pgr) you can see a short video of (not the latest) display menu.
However, this branch is based on the current master (version 2.9.4). The current "next version" development [branch](https://github.com/rancilio-pid/ranciliopid/tree/Alpha-3.0.0) differs very much meanwhile, so an integration will require some effort.

fiendie, 2023-02-10:
The rotary encoder uses 3, so in theory those could be repurposed.

SphaeroX, 2023-07-20:
Pin 14 and 12 free.

These two pins are so called touchpins which means you can make an input by touching them.

Here is a video: https://youtu.be/4YY7TutRrQE?t=196

With this you could also build a control where you can select left and right and if you hold the right button longer you can set a value (flashes then).

LoQue90, 2023-12-24:
there is a wip PR #419 to add a menu on the display with a rotary encoder. It´s allready posible to change the temperature with this menu. Try it out. :)