rancilio-pid/clevercoffee #223: E-Trigger mit Standbyfunktion

murmeltier08, 2022-02-24:
Standbyfunktion mit Timer für Display

tbaumann, 2022-10-18:
If I may hijack this issue. What even is the E-Trigger?
I can't figure it out from the schematics at all.

Is it related to the Rancillio standby mode?

fiendie, 2023-02-09:
@tbaumann: E-Trigger resets the 30 minute timer on Silvia E models (v5 and v6). This also happens when you start a brew or switch on hot water on an unmodified machine.

tml89, 2023-02-21:
Whats the target? To show the E-Trigger timer in the display or implement a standbytimer (e.g. turn pid off after 30min)?

fiendie, 2023-02-21:
That depends on whether you have a Silvia v5+ or if you do, chose to take out the CPU. 

In the former case, standby mode is just a matter of switching off the display and the heater. In the latter case you would need a separate relay to trigger the functions on the CPU.

Either way, I would stop using the word "E-Trigger" and just refer to different implementations of the standby mode.

LoQue90, 2023-12-05:
added from @fiendie with https://github.com/rancilio-pid/clevercoffee/pull/359

closed 2023-12-05 by LoQue90