rancilio-pid/clevercoffee #259: 404 (Not Found) on ESP8266 Webserver

StWiemann, 2022-10-01:
I just upgraded the code of my ESP8266 to get rid of Blynk. 
OTA worked after disabling the fire-wall, I was able to flash the device (multiple times).
Also I got to the point where I could connect to the MCU via access point mode and I entered my WiFi credentials.
Now the MCU starts but when I browse to the IP of the Machine, I only get 404 (Not Found). It seems like the webserver didn't start correctly.
I am able however to navigate to http://<IP>/temperatures and get the temperatures.

I did not yet try to flash it over USB, because I didn't want to open the machine. 

Is there any experience with this kind of behaviour?

I do have the Silvia  E, only PID with a DS18B20 as sensor. 
My config looks like this: 

```
/**
 * @file    userConfig_sample.h
 * @brief   Values must be configured by the user
 * @version 3.0.1 Alpha
 *
 */
#ifndef _userConfig_H
#define _userConfig_H

// firmware version (must match with definitions in the main source file)
#define USR_FW_VERSION    3
#define USR_FW_SUBVERSION 0
#define USR_FW_HOTFIX     2
#define USR_FW_BRANCH     "MASTER"

// List of supported machines
enum MACHINE {
    RancilioSilvia,   // MACHINEID 0
    RancilioSilviaE,  // MACHINEID 1
    Gaggia,           // MACHINEID 2
    QuickMill         // MACHINEID 3
};

/**
 * Preconfiguration
 */

// Machine
#define MACHINEID 1                // see above list of supported machines

// Display
#define OLED_DISPLAY 2             // 0 = deactivated, 1 = SH1106 (e.g. 1.3 "128x64), 2 = SSD1306 (e.g. 0.96" 128x64)
#define OLED_I2C 0x3C              // I2C address for OLED, 0x3C by default
#define DISPLAYTEMPLATE 1          // 1 = Standard display template, 2 = Minimal template, 3 = only temperature, 4 = scale template, 20 = vertical display (see git Handbook for further information)
#define DISPLAYROTATE U8G2_R0      // rotate display clockwise: U8G2_R0 = no rotation; U8G2_R1 = 90°; U8G2_R2 = 180°; U8G2_R3 = 270°
#define SHOTTIMER 1                // 0 = deactivated, 1 = activated 2 = with scale
#define HEATINGLOGO 1              // 0 = deactivated, 1 = Rancilio, 2 = Gaggia
#define OFFLINEGLOGO 1             // 0 = deactivated, 1 = activated
#define BREWSWITCHDELAY 3000       // time in ms that the brew switch will be delayed (shot timer will show that much longer after switching off)
#define VERBOSE 0                  // 1 = Show verbose output (serial connection), 0 = show less

// Connectivity
#define CONNECTMODE 1              // 0 = offline 1 = WIFI-MODE 2 = AP-MODE (not working in the moment)
#define HOSTNAME "Rancilio"
#define PASS "Rancilio"        // default password for WiFiManager
#define MAXWIFIRECONNECTS 5        // maximum number of reconnection attempts, use -1 to deactivate
#define WIFICONNECTIONDELAY 10000  // delay between reconnects in ms

// PID & Hardware
#define ONLYPID 1                  // 1 = Only PID, 0 = PID and preinfusion
#define ONLYPIDSCALE 0             // 0 = off , 1 = OnlyPID with Scale
#define BREWMODE 1                 // 1 = NORMAL preinfusion ; 2 = Scale with weight
#define BREWDETECTION 1            // 0 = off, 1 = Software (Onlypid 1), 2 = Hardware (Onlypid 0), 3 = Sensor/Hardware for Only PID
#define BREWSWITCHTYPE 1           // 1 = normal Switch, 2 = Trigger Switch
#define TRIGGERTYPE HIGH           // LOW = low trigger, HIGH = high trigger relay
#define VOLTAGESENSORTYPE HIGH     // BREWDETECTION 3 configuration
#define PINMODEVOLTAGESENSOR INPUT // Mode INPUT_PULLUP, INPUT or INPUT_PULLDOWN_16 (Only Pin 16)
#define PRESSURESENSOR 0           // 1 = pressure sensor connected to A0; PINBREWSWITCH must be set to the connected input!
#define TEMPLED 1                  // set led pin high when brew or steam set point is within range

// TOF sensor for water level
#define TOF 0                      // 0 = no TOF sensor connected; 1 = water level by TOF sensor
#define TOF_I2C 0x29               // I2C address of TOF sensor; 0x29 by default
#define CALIBRATION_MODE 0         // 1 = enable to obtain water level calibration values; 0 = disable for normal PID operation; can also be done in Blynk
#define WATER_FULL 102             // value for full water tank (=100%) obtained in calibration procedure (in mm); can also be set in Blynk
#define WATER_EMPTY 205            // value for empty water tank (=0%) obtained in calibration procedure (in mm); can also be set in Blynk

// E-Trigger
#define ETRIGGER 0                 // 0 = no trigger (for Rancilio except Rancilio E), 1 = trigger for CPU of Rancilio E
#define ETRIGGERTIME 600           // seconds, time between trigger signal
#define TRIGGERRELAYTYPE HIGH      // LOW = low trigger, HIGH = high trigger relay for E-Trigger

// Brew Scale
#define SCALE_SAMPLES 2                     // Load cell sample rate
#define SCALE_CALIBRATION_FACTOR 3195.83    // Raw data is divided by this value to convert to readable data

/* Pressure sensor
 *
 * measure and verify "offset" value, should be 10% of ADC bit reading @supply volate (3.3V)
 * same goes for "fullScale", should be 90%
 */
#define OFFSET      102            // 10% of ADC input @3.3V supply = 102
#define FULLSCALE   922            // 90% of ADC input @3.3V supply = 922
#define MAXPRESSURE 200

// PlatformIO OTA
#define OTA true                   // true = OTA activated, false = OTA deactivated
#define OTAHOST "Rancilio"     // Name to be shown in ARUDINO IDE Port
#define OTAPASS <password>          // Password for OTA updtates

// MQTT
#define MQTT 0                             // 1 = MQTT enabled, 0 = MQTT disabled
#define MQTT_USERNAME "mymqttuser"
#define MQTT_PASSWORD "mymqttpass"
#define MQTT_TOPIC_PREFIX "custom/Küche."  // topic will be "<MQTT_TOPIC_PREFIX><HOSTNAME>/<READING>"
#define MQTT_SERVER_IP "XXX.XXX.XXX.XXX"   // IP-Address of locally installed mqtt server
#define MQTT_SERVER_PORT 1883

// BLYNK
#define BLYNK 0                                 // 0 = no Blynk , 1 = Blynk
#define AUTH "blynk_auth"
#define BLYNKADDRESS "blynk.clevercoffee.de"    // blynk-server IP-Address
#define BLYNKPORT 8080                          // blynk-server portver

// Grafana
#define GRAFANA 0                  // 2 = custom Grafana 1 = Grafana visualisation (access required), 0 = off (default)

// INFLUXDB
#define INFLUXDB 0                 // 1 = INFLUX enabled, 0 = INFLUX disabled
#define INFLUXDB_URL ""            // InfluxDB server address
#define INFLUXDB_AUTH_TYPE 1       // 1 = API Token , 2 = User/Pass
#define INFLUXDB_API_TOKEN ""
#define INFLUXDB_ORG_NAME ""
#define INFLUXDB_USER ""
#define INFLUXDB_PASSWORD ""
#define INFLUXDB_DB_NAME "coffee"  // InfluxDB bucket name
#define INFLUXDB_INTERVAL 5000     // Send interval in milliseconds

// PID Parameters (not yet in Web interface)
#define EMA_FACTOR 0.6             // Smoothing of input that is used for Tv (derivative component of PID). Smaller means less smoothing but also less delay, 0 means no filtering
#define BREWPID_DELAY 10           // delay until enabling PID controller during brew (no heating during this time)

// Backflush values
#define FILLTIME 3000              // time in ms the pump is running
#define FLUSHTIME 6000             // time in ms the 3-way valve is open -> backflush
#define MAXFLUSHCYCLES 5           // number of cycles the backflush should run, 0 = disabled

// Pin Layout
#define ONE_WIRE_BUS 2             // Temp sensor pin
#define PINPRESSURESENSOR 99       // Pressuresensor 0: A0 (ESP8266), >0 ONLY ESP32
#define PINVALVE 12                // Output pin for 3-way-valve
#define PINPUMP 13                 // Output pin for pump
#define PINHEATER 14               // Output pin for heater
#define PINVOLTAGESENSOR 15        // Input pin for voltage sensor
#define PINETRIGGER 16             // PIN for E-Trigger relay
#define PINBREWSWITCH 0            // 0: A0 (ESP8266) ; >0 : DIGITAL PIN, ESP32 OR ESP8266: ONLY USE PIN15 AND PIN16!
#define PINSTEAMSWITCH 17          // STEAM active
#define LEDPIN    18               // LED PIN ON near setpoint
#define OLED_SCL 5                 // Output pin for display clock pin
#define OLED_SDA 4                 // Output pin for display data pin
#define HXDATPIN 99                // weight scale PIN
#define HXCLKPIN 99                // weight scale PIN
#define SCREEN_WIDTH 128           // OLED display width, in pixels
#define SCREEN_HEIGHT 64           // OLED display height, in pixels

// Historic (no settings)
#define TEMPSENSOR 1               // 2 = TSIC306, 1 = DS18B20

// Check BrewSwitch
#if (defined(ESP8266) && ((PINBREWSWITCH != 15 && PINBREWSWITCH != 0 && PINBREWSWITCH != 16 )))
  #error("WRONG Brewswitch PIN for ESP8266, Only PIN 15 and PIN 16");
#endif

// defined compiler errors
#if (PRESSURESENSOR == 1) && (PINPRESSURESENSOR == 0) && (PINBREWSWITCH == 0)
  #error Change PINBREWSWITCH or PRESSURESENSOR!
#endif

#endif

```

Thank you!

kjyv, 2022-10-02:
You'll need to also run the PlatformIO tasks "Upload Filesystem" and possibly also "Erase Flash" (but you can first try without it)

StWiemann, 2022-10-05:
Thank you that helped to get the web-page. In addition to make it work I had to do this for anybody having the same problem:
https://github.com/platformio/platform-espressif8266/issues/263#issuecomment-1081177511

Unfortunately the device does not save the set-temperature and PID-values now. Haven't figured that one out yet. But the inital Problem ist solved.

closed 2022-10-05 by StWiemann

kjyv, 2022-10-05:
We haven't seen this before I think, which OS is this on? Saving not being persistent is usually because "Erase Flash" was not used (only works via USB)

StWiemann, 2022-10-05:
Win 11, VSC 1.71.2, PlatformIO Core 6.1.4 Home 3.4.3
I assumed that the switch to littlefs already did erase the flash, but I guess I'll have to give USB a try then, thank you!