rancilio-pid/clevercoffee #260: chat.rancilio-pid.de

tbaumann, 2022-10-01:
There are a bunch of references to the defunct site https://chat.rancilio-pid.de/ but that system seems offline.

Is that a permanent thing or is that supposed to come back? If not I guess some docu references should be cleaned.

kjyv, 2022-10-02:
It is working quite well for me, so might have been a temporary issue

murmeltier08, 2022-10-09:
work by me

closed 2022-10-09 by murmeltier08

tbaumann, 2022-10-17:
Definitely not working. Is there perhaps some geo-restriction? I'm accessing from Morocco.

tbaumann, 2022-10-17:
I'm routed all the way into hetzner. If the service is running then something very close to the target is blackholing me

```
$ tracepath 138.201.213.140
 1?: [LOCALHOST]                      pmtu 1500
 1:  _gateway                                              2.159ms 
 1:  _gateway                                              2.039ms 
 2:  _gateway                                              2.007ms pmtu 1492
 2:  41.251.0.1                                           10.649ms 
 3:  adsl-178-65-192-81.adsl2.iam.net.ma                   9.636ms 
 4:  adsl-177-65-192-81.adsl2.iam.net.ma                   9.309ms 
 5:  static-81-192-32-46.iam.net.ma                        8.475ms 
 6:  213.144.176.172                                      32.812ms asymm 10 
 7:  bundle-ether306.madtr6.madrid.opentransit.net        32.213ms asymm 10 
 8:  ae9.cr6-fra2.ip4.gtt.net                             60.222ms asymm 14 
 9:  ip4.gtt.net                                          70.098ms asymm 14 
10:  core23.fsn1.hetzner.com                              75.729ms asymm 15 
11:  ffm-b5-link.ip.twelve99.net                          60.507ms asymm 12 
12:  hetzner-ic326013-ffm-b5.ip.twelve99-cust.net         78.310ms asymm 14 
13:  core23.fsn1.hetzner.com                              82.732ms asymm 15 
14:  ex9k2.dc8.fsn1.hetzner.com                           85.249ms asymm 16 
15:  no reply
16:  no reply
17:  no reply
18:  no reply
19:  no reply
20:  no reply
```

kjyv, 2022-10-18:
Could you try a VPN to check if that's the issue?

tbaumann, 2022-10-18:
> Could you try a VPN to check if that's the issue?

Yes, just did. It works on VPN. So please guys respect the Inter in Internet. ;)

kjyv, 2022-10-18:
Ok good. I don't think it has anything to do with us, it's more likely that some IP ranges are blocked in Marocco adn that you'll have to use a VPN

tbaumann, 2022-10-18:
> Ok good. I don't think it has anything to do with us, it's more likely that some IP ranges are blocked in Marocco and that you'll have to use a VPN

Nope, I get routed all the way to Hetzner. It's either Hetzner (unlikely) or the Mattermost instance or it's firewall.

tbaumann, 2022-10-18:
It's not the mattermost service itself because all IP traffic gets blocked not just the web ports.

kjyv, 2022-10-18:
If I do a traceroute on the ip, I get pretty much the same result, the last hop is ex9k2.dc8.fsn1.hetzner.com.
I have not set up the server, so can't say that nothing is blocked.

tbaumann, 2022-10-18:
PS: The issue is across different local providers

tbaumann, 2022-10-18:
> If I do a traceroute on the ip, I get pretty much the same result, the last hop is ex9k2.dc8.fsn1.hetzner.com. I have not set up the server, so can't say that nothing is blocked.

Can we make the person responsible for this somewhow aware? Would be real nice.