rancilio-pid/clevercoffee #269: Pre select PID value table from Handbook

cottec, 2022-10-19:
Might be easy for quick setup of standard values depending on machine type

kjyv, 2022-10-19:
Do you mean to suggest a preset selection UI on the parameters page? There have been some ideas about this already but no one has taken the time to work on it - feel free to start ;)

cottec, 2022-10-19:
Yeah or automatically load them depending on machine type in userconfig.h
Probably makes more sense to hardcode the standard settings for each machine type?
`https://github.com/rancilio-pid/ranciliopid/blob/9bb3816a4f0c485dc99e01975b2b5ae1dbef0629/src/rancilio-pid.h#L16`

kjyv, 2022-10-19:
We are going to remove the config file completely in the near future, so pre-selecting the machine type won't be necessary anymore. I guess a first time setup-page would be nice asking for features, machine etc.