rancilio-pid/clevercoffee #278: compile with 1 = DS18B20 sensor fails

creisey, 2022-11-07:
**ompiling .pio\build\nodemcuv2_usb\src\rancilio-pid.cpp.o
src\rancilio-pid.cpp:307:5: error: 'OneWire' does not name a type
     OneWire oneWire(PINTEMPSENSOR);         // Setup a OneWire instance to communicate with OneWire
     ^
src\rancilio-pid.cpp:309:5: error: 'DallasTemperature' does not name a type
     DallasTemperature sensors(&oneWire);
     ^
src\rancilio-pid.cpp:310:5: error: 'DeviceAddress' does not name a type
     DeviceAddress sensorDeviceAddress;      // arrays to hold device address
     ^
src\rancilio-pid.cpp: In function 'void refreshTemp()':
src\rancilio-pid.cpp:599:13: error: 'sensors' was not declared in this scope
             sensors.requestTemperatures();
             ^
src\rancilio-pid.cpp: In function 'void setup()':
src\rancilio-pid.cpp:1872:9: error: 'sensors' was not declared in this scope
         sensors.begin();
         ^
src\rancilio-pid.cpp:1873:28: error: 'sensorDeviceAddress' was not declared in this scope
         sensors.getAddress(sensorDeviceAddress, 0);
                            ^
In file included from src\rancilio-pid.cpp:33:0:
src\languages.h: At global scope:
src\languages.h:22:20: warning: 'langstring_offlinemod' defined but not used [-Wunused-variable]
 static const char *langstring_offlinemod = "Offlinemodus";
                    ^
src\languages.h:23:20: warning: 'langstring_waterempty' defined but not used [-Wunused-variable]
 static const char *langstring_waterempty = "Wasser leer";
                    ^
src\languages.h:25:20: warning: 'langstring_wifirecon' defined but not used [-Wunused-variable]
 static const char *langstring_wifirecon = "Wifi reconnect:";
                    ^
src\languages.h:26:20: warning: 'langstring_connectwifi1' defined but not used [-Wunused-variable]
 static const char *langstring_connectwifi1 = "1: Connect Wifi to:";
                    ^
src\languages.h:27:20: warning: 'langstring_nowifi' defined but not used [-Wunused-variable]
 static const char *langstring_nowifi[] = {"No ", "WIFI"};
                    ^
src\languages.h:29:20: warning: 'langstring_error_tsensor' defined but not used [-Wunused-variable]
 static const char *langstring_error_tsensor[] = {"Fehler, Temp: ", "Temp.-Sensor ueberpruefen!"};
                    ^
src\languages.h:32:20: warning: 'langstring_bckffinished' defined but not used [-Wunused-variable]
 static const char *langstring_bckffinished[] = {"Backflush beendet", "Bitte Bruehschalter abschalten..."};
                    ^
src\languages.h:33:20: warning: 'langstring_bckfactivated' defined but not used [-Wunused-variable]
 static const char *langstring_bckfactivated[] = {"Backflush aktiviert", "Bruehschalter betaetigen ..."};
                    ^
src\languages.h:34:20: warning: 'langstring_bckfrunning' defined but not used [-Wunused-variable]
 static const char *langstring_bckfrunning[] = {"Backflush aktiv:", "seit"};
                    ^
*** [.pio\build\nodemcuv2_usb\src\rancilio-pid.cpp.o] Error 1
======================================================================================================================== [FAILED] Took 6.09 seconds ========================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
nodemcuv2_usb  FAILED    00:00:06.088
=================================================================================================================== 1 failed, 0 succeeded in 00:00:06.088 ===================================================================================================================

 *  The terminal process "C:\Users\superuser\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'nodemcuv2_usb'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. **

xTraXx, 2022-11-09:
The problem is in rancilio-pid.cpp

`#if TEMPSENSOR == 1
    #include <OneWire.h>
    #include <DallasTemperature.h>  // Library for dallas temp sensor
#endif`

Should be after the "userConfig.h" import. Because the TEMPSENSOR setting is not know before.

creisey, 2022-11-09:
grande, so i could compile the firmware with this order:

[image]
`
/**
 * @file rancilio-pid.cpp
 *
 * @brief Main sketch
 *
 * @version 3.1.0 Master
 */

// Firmware version
#define FW_VERSION    3
#define FW_SUBVERSION 1
#define FW_HOTFIX     0
#define FW_BRANCH     "MASTER"

// Libraries
#include <ArduinoOTA.h>
#if TOF == 1
    #include <Adafruit_VL53L0X.h>   // for ToF Sensor
#endif

#include <WiFiManager.h>
#include <InfluxDbClient.h>
#include <PubSubClient.h>
#include <U8g2lib.h>            // i2c display
#include "TSIC.h"               // old library for TSIC temp sensor
#include <ZACwire.h>            // new TSIC bus library
#include "PID_v1.h"             // for PID calculation

// Includes
#include "icon.h"               // user icons for display
#include "languages.h"          // for language translation
#include "Storage.h"
#include "ISR.h"
#include "debugSerial.h"
#include "userConfig.h"         // needs to be configured by the user
#if TEMPSENSOR == 1
    #include <DallasTemperature.h>  // Library for dallas temp sensor
#endif
#include "rancilio-pid.h"
`

creisey, 2022-11-09:
https://github.com/rancilio-pid/clevercoffee/commit/c7133d54b49e1d39dc2b11421e5213cf3a473a82

creisey, 2022-11-09:
thx a lot

PR #285 (2022-11-23): Pull include of userConfig.h forward
To fix build issues due to TOF and TEMPSENSOR being undefined, the include of userConfig.h is pulled forward.

This fixes #284 and fixes #278.
closed 2022-11-09 by creisey