rancilio-pid/clevercoffee #282: MQTT Topic for available

Chillkroetchen, 2022-11-14:
Is it reasonable to have a seperate MQTT topic for machine availability?
For example: `kitchen/clevercoffee/available/` and set this to 1 as soon as machine goes online.
If it's not usually done that way, which approach would make more sense?

Chillkroetchen, 2022-11-14:
I set Home Assistant to send a push notification to my phone if the temperature is between 94 and 96 degrees for 5min. I will have to do some more testing but until now it works quite well without sending duplicate notifications.

kjyv, 2022-12-11:
I guess because mqtt does not show if the machine is currently sending data or if the values are old and the machine is actually turned off?

kjyv, 2022-12-11:
Uptime is easy, available not sure - it would mean sending some kind of ping or maybe there is built in functionality for that in the mqtt library

Chillkroetchen, 2022-12-11:
It could be used to set certain automations in homeassistant that are only run if an availability topic is set to true. Only to use can be unreliable at times because I don't know a way to differ between live and outdated messages.
Additionally if you display temperature in homeassistant, it will show the last temperature that the machine transmitted before disconnecting.
A special availability topic that is set _true_ when offline with a last will message _false_ would be a great help here.

kjyv, 2022-12-11:
But how would that work? It will simply stay at its last value as well, so it will always appear as available. Maybe sending the current time every second or so would be a better way to do this

Chillkroetchen, 2022-12-11:
http://www.steves-internet-guide.com/mqtt-last-will-example/

If you write available _true_ you can send that message with a last will message _false_
In case of unexpected connection timeout the broker will publish the last will message in the machines name.

LoQue90, 2022-12-12:
Availability with last will would be really handy in home assistant.

Nilaik, 2022-12-28:
I created a merge request that implements an available topic with last will: https://github.com/rancilio-pid/clevercoffee/pull/295

This works fine for my Home Assistant setup.

Nilaik, 2022-12-28:
The configuration in Home Assistant looks like this:

```
  - name: Rancilio Silvia Temperature
    unique_id: sensor.silvia_temp
    availability:
      - payload_available: online
        payload_not_available: offline
        topic: espresso/rancilio.silvia/available
    state_topic: espresso/rancilio.silvia/temperature
    state_class: measurement
    device_class: temperature
    unit_of_measurement: °C
```

Home Assistant sets the sensor to not available if the last availability topic received contains the palyoad_not_available content. As long as the last time the availability topic was received contained the payload_available the sensor is marked available.

More information can be found here: https://www.home-assistant.io/integrations/sensor.mqtt/#availability

murmeltier08, 2022-12-31:
I think this is solved with the merge of #295

closed 2022-12-31 by murmeltier08