rancilio-pid/clevercoffee #293: MQTT-Topic for machine state

gang-himself, 2022-12-23:
I'd love to have the machine-state as mqtt-topic, as this would help to make some automations or statistics in for example homeassistant.
I hope this should be quite easy, as the machine-state is already printed in the debug-console.

Thanks,
Wolfgang

murmeltier08, 2023-01-04:
We will add it later for the esp32 build.

gang-himself, 2023-01-04:
Is there a chance to get it also in the ESP8266-Build, as this is the one I‘ve used…

murmeltier08, 2023-01-04:
i will check it, if it possible

Nilaik, 2023-01-04:
I created  a pull request how it could be done: https://github.com/rancilio-pid/clevercoffee/pull/300

However, if state changes appear fast it will not publish all of them because the function writeSysParamsToMQTT is not called only at the end of the loop. Sending the machine state could also be added to printMachineState but that felt kind of weird since this function only contains debug prints.

murmeltier08, 2023-01-04:
Thank you, PR is merged

closed 2023-01-04 by murmeltier08

gang-himself, 2023-01-08:
Thanks, works as expected.