rancilio-pid/clevercoffee #319: Brew detection not detecting that the brewing already stopped

niklasarnitz, 2023-01-27:
It ran until 200s, I then rebooted the machine.

PID+, so it's temperature based brew detection.

V3.1.1

kjyv, 2023-01-27:
OnlyPID+ is using hardware to detect the brew switch state. Do you mean OnlyPID with Software brew detection?
You might have to fine tune its parameters to make the detection work better in you case, see the manual.

niklasarnitz, 2023-01-27:
[image]

kjyv, 2023-02-04:
So, I've tried to reproduce this with the current version 3.1.2 and could not see anything wrong. First of all, software brew detection does not detect that the brew stopped (it can't really know when that is because the temperature does not do anything obvious when the brew stops), it simply stops after the set time - you have set 120 seconds. If I wait that long, the counter disappears and everything is back to the state it was before. You probably don't want to do any brews that are that long, so 40 or 50 should be enough. With OnlyPID+ or changing the brew switch to a trigger, you can't expect to see accurate time measurements, also the start is always a bit delayed. The only purpose of this is to set different PID parameters during brew, but you also are likely to not need that (depends on the machine type).

niklasarnitz, 2023-02-05:
Okay, that makes sense.
Sorry for the confusion!

closed 2023-02-05 by niklasarnitz