rancilio-pid/clevercoffee #323: After first time setting WiFi credentials and connecting, Webserver is not reachable

kjyv, 2023-01-29:


LoQue90, 2023-04-16:
thats a wifimanger/wifiserver bug. the wifimanger opens a port for the captive portal.
when it´s done, it closes the port, then the asyncwebserver trys to reopen the same port. That is not possible.
see: https://github.com/tzapu/WiFiManager/issues/1579

patelabhi23, 2023-05-01:
@kjyv were you able to get solution of this?

> When setting up a new machine, the wifi credentials can be entered and the esp connects to it and displays an IP in the the log. However, connecting to it and showing the webinterface is only possible when restarting the ESP

kjyv, 2023-05-29:
@patelabhi23 No, and I guess we can't really do anything on this side if it's a bug in WiFiManager

khoint2101, 2023-06-12:
> When setting up a new machine, the wifi credentials can be entered and the esp connects to it and displays an IP in the the log. However, connecting to it and showing the webinterface is only possible when restarting the ESP

I have a idea to fix your issue. You won't have to hard reset anymore. That mean restart by code.
- The first, the reason is WM has samilar port with you ESPAsyncWebserver ( port 80 http), after you reset wifi in4 and reconfig wifi, the Webconfigportal don't return port 80 ( or ESP port has hard error). Therefore when ESPAsyncWebserver start will error bind -8 error. You must hard restart ESP32 and This annoys you =))
- I have using WM with NonBlocking Mode:

```
#include <WiFi.h>
#include <ESPAsyncWebServer.h>
#include <WiFiManager.h>
#include <AsyncTCP.h>
#include <FS.h>

AsyncWebServer server(80);
WiFiManager wm; // global wm instance

bool shouldRestart = false; // flag to track if restart is needed

void setup(){
   WiFi.mode(WIFI_STA);
   wm.setConfigPortalBlocking(false);
   Serial.setDebugOutput(true);
    // delay(3000);   // if you won't change port, you must hard restart
    // wm.setHttpPort(8080); // set another port for WM because https://github.com/rancilio-pid/clevercoffee/issues/323
    Serial.println("\n Starting");
    pinMode(TRIGGER_PIN, INPUT);
    wm.setConfigPortalBlocking(false);
   
    // set dark theme
    wm.setClass("invert");
    wm.setConfigPortalTimeout(60); // auto close configportal after n seconds
    bool res;
  
    res = wm.autoConnect("StartConfig_WF"); // password protected ap

    if (!res)
    {
        Serial.println("Failed to connect or hit timeout");
        // ESP.restart();
        shouldRestart = true;     // flag set true
    }
    else
    {
        // if you get here you have connected to the WiFi   
        Serial.println("connected...yeey :)"); 
        initFS();
        initWebSocket();
        server.on("/", HTTP_GET, [](AsyncWebServerRequest *request)         //start Webserver UI
                  { request->send(SPIFFS, "/index.html", "text/html"); });

        // Send a GET request to <IP>/get?message=<message
        // Send a POST request to <IP>/post with a form field message set to <message>

        server.onNotFound(notFound);
        server.serveStatic("/", SPIFFS, "/");
        server.begin();  // begin server
    }
}
void checkButton()
{
    // check for button press
    if (digitalRead(TRIGGER_PIN) == LOW)
    {
        // poor mans debounce/press-hold, code not ideal for production
        delay(50);
        if (digitalRead(TRIGGER_PIN) == LOW)
        {
            Serial.println("Button Pressed");
            // still holding button for 3000 ms, reset settings, code not ideaa for production
            delay(3000); // reset delay hold
            if (digitalRead(TRIGGER_PIN) == LOW)
            {
                Serial.println("Button Held");
                Serial.println("Erasing Config, restarting");
                wm.resetSettings();
                ESP.restart();
            }
        }
    }
}

void checkWifi_config()    // if wifi connected AND flag config true, restart esp
{
    // nếu wifi config thành công sau khi reset thông tin wifi thì sẽ reset để mở lại port 80 cho webserver
    if ((WiFi.status() == WL_CONNECTED) && shouldRestart == true)
    {
        ESP.restart();
    }
}

void loop(){
    wm.process();
    checkButton();
    checkWifi_config();  // restart if reset wifi and config again

}
```
[image]
[image]

patelabhi23, 2023-06-13:
@khoint2101 Thanks for the reply. I tried this code but it has some issues.

1) You have mentioned about  `initWebSocket();` but isn't defined
2) Assigning a value to the `trigger_pin` should be optional, as it is not necessary for every instance that a person clicks on the pin. In cases where it is necessary, it is more convenient to manually restart the system from the power supply rather than relying on this functionality.
3) ESP.restart didn't work in my case

LoQue90, 2023-06-13:
automatic restart after wifi setup done, sounds great.
but using the boot button for reset wifi creds, im not sure. Sounds for me like something that adds a lot of support, because people accidentiliy push that button.
But having a button on the website that erase the current wifi creds and bring up wifi manager on the next boot would be nice

khoint2101, 2023-06-13:
@LoQue90 

> automatic restart after wifi setup done, sounds great. but using the boot button for reset wifi creds, im not sure. Sounds for me like something that adds a lot of support, because people [...]

`because people accidentiliy push that button` long press for 3 seconds or 5 seconds BOOT BUTTON will prevent users from accidentally pressing the wrong button. Your idea of ​​creating a button right on the web config is really great!

LoQue90, 2023-12-05:
@khoint2101 
are you interestet to bring your idea as an pr?

closed 2025-07-23 by LoQue90