rancilio-pid/clevercoffee #324: Make temperature offset proportional

krisraich, 2023-02-01:
Hi

I'm using firmware v3.1.2 on my Rancilio where I've set the temperature offset to 10°.
It would be nice, if this offset would not just be subtracted from the actual temperature reading.
The reason is, that when the machine starts, it displays a temperature that's way lower than the room temperature. 

- Ts = set temperature
- Td = temperature displayed
- Tb = temperature boiler
- Δ = offset
- p = power (e.g. 2)

I would suggest instead of using  Td = Tb - Δ
Maybe something like Td = Tb - ( ( ( Tb - Δ ) / Ts ) ^ p ) * Δ

[image]

Maybe you can also implement this in your ESP8266-MASTER baranch?

kjyv, 2023-02-01:
You have a very cold room if your temperature is below 10°...
This offset is of course only valid when your machine is heated up, it is supposed to describe the offset between boiler and grouphead in the steady state where the heat loss to the ambience is the same as the heat that is transferred from the boiler to the group. I'm not quite sure what your equation is modeling instead?

krisraich, 2023-02-01:
That's the point. With the proposed equation, the offset is gradually applied to the displayed temperature. 
E.g.: Boiler temp: 24° -> displayed temp: 24°, 103° -> 93°
Also, it is closer to the a real temperature gradient, since the temperature difference between group head and boiler increases with boiler temperature.

No its not an issue.. its a nice to have. 

`With your calculation it would always be something that is not based on any actual physical temperature it seems.` Well that's also the case with an fixed offset. It only is accurate when the set temperature is reached.
As shown in my screenshot, when the set temperature is reached, the exact same offset is subtracted as with the current approach. 
So if:

- Tb < (Ts + Δ) -> the subtracted offset is smaller than Δ
- Tb = Ts + Δ -> the offset = Δ
- Tb > (Ts + Δ) -> the subtracted offset is greater than Δ

kjyv, 2023-02-01:
Ah sorry, I overlooked that you end up on exactly the set offset when Ts is reached.
Though when you switch the machine on, the boiler and group are at the same temperature for only a short time. As soon as the boiler heats up, the difference grows quickly with the increasing boiler temperature and the group only slowly warms up while the boiler is already almost at the setpoint. So probably you'll get differences of 30° or more for a while. So I guess the offset gradient is basically showing the boiler temp for most of the time until the boiler is mostly heated up and then smoothly fades over to the approximate group temperature - not sure if this is also confusing to some people?
Also there might be some implementation details that make this a bit more complicated as there is code to detect sensor errors that would have to take the current offset into account.

kjyv, 2023-02-01:
I can't decide alone if we add this or not but feel free to join the chat (https://chat.rancilio-pid.de/) and propose the idea there. Ideally also make a PR for it ;)

kjyv, 2023-02-01:
Oh and it would actually be really cool if we could train an estimator for the group temp - we would need to first get some data for each machine type though

murmeltier08, 2023-02-06:
In my personal point of view is the offset in general more a "visual hack". We only know the boiler temperature and not the temperature in the group head. If you cool down the boiler or generate steam the temperature will change and every offset calculation is not possible.

kjyv, 2023-02-07:
Sure it's a visual thing only but it's also not very helpful currently to see the boiler temperature if there are significant differences to the actual water temperature that hits the coffee. This suggestion doesn't really change that but tries to make the visuals a bit nicer though.

LoQue90, 2023-02-07:
but the offset will not give you a valid temp of your grouphead. it´s just guesswork.
why is this nicer showing a temp that is maybe right, instead of showing something we actully have a sensor for.

i set my temp of the boiler based on taste in the cup.
so i know setpoint 98°C is perfect for my standard bean for my capuccino.

justmy2cents :)

LoQue90, 2023-02-07:
> I don't find an offset particularly useful either. But now that we have one anyway I don't it would hurt to do this kind of curve fitting so it gives more realistic readings at every stage. As long as it's all turned off by default so I don't have to ;)

sounds good!

kjyv, 2023-02-07:
Sure, it's always a matter of personal preference and that's why its optional. I don't think any of the cases (static offset, some gradient or no offset) are giving in any way worse information than the others and all need to take heat up time, flushing and steaming and the final taste in the cup into account. With the offset you however have the option of adjusting the visible temperature (at the right time and conditions) to a range you can compare with other brew methods and machines. You don't actually brew your espresso with 98°, so it just looks wrong to some people including me.

kjyv, 2023-02-07:
With the sensor on top, you don't even measure the temperature of the water that's leaving the boiler btw., you will actually see a few degress above the actual temperature ;)

LoQue90, 2023-02-07:
In the end I don't care at which exact temp I brew.
But the sensor will give me a referenz.
Anything else on top will make me wonder what I come from.
Default off will make me total happy. :)

janv, 2023-02-23:
@krisraich  You can use LateX here :)

```md
$T_s$ = set temperature
$T_d$ = temperature displayed
$T_b$ = temperature boiler
$\Delta$ = offset
$p$ = power (e.g. 2)
I would suggest instead of using $T_d = T_b - \Delta$
Maybe something like

$$ T_d = T_b - \left( \frac{ T_b - \Delta }{T_s} \right) ^ p  \times \Delta $$
```

gives:

> 
$T_s$ = set temperature
$T_d$ = temperature displayed
$T_b$ = temperature boiler
$\Delta$ = offset
$p$ = power (e.g. 2)
I would suggest instead of using $T_d = T_b - \Delta$
Maybe something like

$$ T_d = T_b - \left( \frac{ T_b - \Delta }{T_s} \right) ^ p  \times \Delta $$

fiendie, 2023-02-23:
I would prefer C-Code, we're not writing an academic paper ;)

closed 2023-09-30 by murmeltier08

SphaeroX, 2024-02-25:
double calculateDisplayTemperature(double T_b, double T_s, double Delta, double p) {
  double T_d = T_b - pow((T_b - Delta) / T_s, p) * Delta;
  return T_d;
}

void setup() {
  Serial.begin(115200);
}

void loop() {
  // Example Data
  double T_s = 100.0; // Set temperature
  double T_b = 105.0; // Boiler temperature
  double Delta = 10.0; // Offset
  double p = 2.0; // Power

  double T_d = calculateDisplayTemperature(T_b, T_s, Delta, p);

  Serial.println(T_d);

  delay(1100);
}