rancilio-pid/clevercoffee #353: WiFi Manager problems

nheilemann, 2023-05-13:
I am trying to install version 3.1.3 on my esp8266.
After flashing, building filesystem and uploading it, building the code and uploading it, the serial monitor shows this message in a loop:

`*wm:[1] AutoConnect 
*wm:[2] Setting Hostnames:  silvia
*wm:[2] Setting WiFi hostname 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connect Wifi, ATTEMPT # 1 of 3
*wm:[1] No wifi saved, skipping 
*wm:[2] Connection result: WL_NO_SSID_AVAIL
*wm:[1] Connect Wifi, ATTEMPT # 2 of 3
*wm:[1] No wifi saved, skipping 
*wm:[2] Connection result: WL_NO_SSID_AVAIL
*wm:[1] Connect Wifi, ATTEMPT # 3 of 3
*wm:[1] No wifi saved, skipping 
*wm:[2] Connection result: WL_NO_SSID_AVAIL
*wm:[1] AutoConnect: FAILED 
*wm:[2] Starting Config Portal 
*wm:[2] AccessPoint set password is VALID 
*wm:[2] Disabling STA 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  silvia
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] WiFi Scan ASYNC started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
*wm:[2] Portal Timeout In 60 seconds

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:133 __yield

>>>stack>>>

ctx: sys
sp: 3ffff5a0 end: 3fffffb0 offset: 0000
3ffff5a0:  40233018 00000000 00001388 4022b820  
3ffff5b0:  000000fe 00000000 00000000 00000000  
3ffff5c0:  00000000 00000000 00000000 3fff2164  
3ffff5d0:  00000000 40225b50 3fff187c 3ffe9632  
3ffff5e0:  00000000 3fff197c 3fff187c 40228ba6  
3ffff5f0:  3fff1ff4 3fff197c 3ffff62c 40228c03  
3ffff600:  3ffff638 3ffff620 3fff187c 40228481  
3ffff610:  00000000 3fff197c 3fff187c 40222f45  
3ffff620:  63657300 73646e6f 80000000 20003000  
3ffff630:  3fff3e00 80000000 73203000 6e6f6365  
3ffff640:  80007364 73203000 6e6f6365 80007364  
3ffff650:  3ffe962b 3ffff680 3ffff68c 3ffe933e  
3ffff660:  3fff19a4 3fff1ed5 3fff187c 3ffe933e  
3ffff670:  00000001 3fff1ed5 3fff187c 40223113  
3ffff680:  00000000 3fff1ecd 80ff0af8 40201d00  
3ffff690:  3fff1ecd 80000006 3ffff6c0 40226d65  
3ffff6a0:  00000001 3ffe85e0 3fff187c 40226d84  
3ffff6b0:  3ffe962b 3ffe9632 00000001 4021b166  
3ffff6c0:  766c6900 00006169 80ff0af8 3ffe97fc  
3ffff6d0:  00000001 3ffe85e0 3fff187c 4020c7e5  
3ffff6e0:  00000005 00000007 3fff0af8 00000000  
3ffff6f0:  00000007 0000000f 3fff0af8 40202027  
3ffff700:  00000007 00000007 3fff0af8 4020954c  
3ffff710:  00000fe8 000001fd 000001fd 40100cff  
3ffff720:  0000002f 000000a2 3fff0af8 3ffe97fc  
3ffff730:  00000000 00000001 00000020 40100f87  
3ffff740:  3fffff38 3ffff780 3fff0af8 402095ac  
3ffff750:  3fffff38 3ffff780 3fffff5c 40226d65  
3ffff760:  3fffff38 3ffff780 3fff0af8 3ffe97fc  
3ffff770:  00000001 00000001 3fff0af8 4020ebe2  
3ffff780:  5f444900 fe004e4f 80efeffe 3fff3700  
[... 184 lines cut ...]
0x4020a518 in std::_Function_base::_Base_manager<setup::{lambda()#17}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#17}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40231d64 in std::_Function_handler<bool (), setup::{lambda()#17}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a550 in std::_Function_base::_Base_manager<setup::{lambda()#18}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#18}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40231d68 in std::_Function_handler<bool (), setup::{lambda()#18}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a588 in std::_Function_base::_Base_manager<setup::{lambda()#19}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#19}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40209e84 in std::_Function_handler<bool (), setup::{lambda()#19}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a5c0 in std::_Function_base::_Base_manager<setup::{lambda()#20}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#20}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40209e94 in std::_Function_handler<bool (), setup::{lambda()#20}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a5f8 in std::_Function_base::_Base_manager<setup::{lambda()#21}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#21}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40209ea4 in std::_Function_handler<bool (), setup::{lambda()#21}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a630 in std::_Function_base::_Base_manager<setup::{lambda()#22}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#22}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40231d6c in std::_Function_handler<bool (), setup::{lambda()#22}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a668 in std::_Function_base::_Base_manager<setup::{lambda()#23}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#23}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40231d70 in std::_Function_handler<bool (), setup::{lambda()#23}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a6a0 in std::_Function_base::_Base_manager<setup::{lambda()#24}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#24}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40231d74 in std::_Function_handler<bool (), setup::{lambda()#24}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a6d8 in std::_Function_base::_Base_manager<setup::{lambda()#25}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#25}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40231d78 in std::_Function_handler<bool (), setup::{lambda()#25}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a70c in std::_Function_base::_Base_manager<setup::{lambda()#26}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#26}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40231d7c in std::_Function_handler<bool (), setup::{lambda()#26}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4022854c in loop_wrapper() at core_esp8266_main.cpp:?

--------------- CUT HERE FOR EXCEPTION DECODER ---------------`

Any ideas? Version 3.10 works fine with the same setup.

LoQue90, 2023-05-15:
that is a known issue, see #302

nheilemann, 2023-05-19:
Thanks, good to know.

For everyone with the same problem: I have figured out a workaround, until the problem is solved.

1. Ease flash
2. Build filesystem
3. Upload filesystem
4. Build the files with "#define OLED_DISPLAY 0" in the userConfig.h
5. Upload files
6. Start the esp and configure your WiFi with the wifi-manager
7.  Build the files with "#define OLED_DISPLAY 2" in the userConfig.h
8. Upload files
--> WiFi and Display are working

I know that this is only a temporary solution. But as long as we have to wait for a real fix, this should be a nice workaround.