rancilio-pid/clevercoffee #366: Coldstart interrupted from shot

LoQue90, 2023-08-13:
Starting a shot during coldstart interrupts the coldstart.
When shot is finished or aborted mashine is switching into normal pid mode, instead of going back to coldstart.
At this point setpoint isn´t reached.

kjyv, 2023-09-17:
Do you have step by step instructions on how to get this reproducibly?

kjyv, 2023-09-17:
It seems from the code this can only happen if the temperature is betweeen the setpoint and (setpoint -1)

LoQue90, 2023-09-20:
I´m using hardware brew decetion with trigger (onlypid 0; brewdetection 2)
But I didn´t checked if the mashine is going back into coldstart mashine state and just don´t show the fullscreen coldstart screen.
I did assumed that the mashine state defines the display state.
gonna check that with the serial monitor

kjyv, 2023-09-20:
I think I can reproduce it if what you mean is basically that the heating up logo is not displayed again after the brew timer was shown, instead it shows the normal screen that would be shown when heatup has finished instead. Is that what you mean?

kjyv, 2023-09-20:
Ah okay, yeah this is just not implemented. The brew detection is already active at some point during the starting modes and when the brew end is detected, it always goes to normal mode. Probably it makes sense not to show the brew timer during first heating (or remember where we came from)

kjyv, 2023-10-01:
Some weird stuff is happening when I check the states during cold start and then flipping the brew switch:
[00:01:41] new machineState: PID Offline -> Cold Start
[00:01:41] new PID-Values: P=55.0  I=0.8  D=715.0
[00:02:16] new machineState: Cold Start -> Steam
[00:02:16] new PID-Values: P=150.0  I=0.0  D=0.0
[00:02:16] new machineState: Steam -> Cool Down
[00:02:16] new PID-Values: P=50.0  I=0.0  D=1000.0
[00:02:16] new machineState: Cool Down -> PID Normal
[00:02:16] new PID-Values: P=55.0  I=0.8  D=715.0
[00:02:16] new machineState: PID Normal -> Steam
[00:02:16] new PID-Values: P=150.0  I=0.0  D=0.0
[00:02:16] new machineState: Steam -> Cool Down
[00:02:16] new PID-Values: P=50.0  I=0.0  D=1000.0
[00:02:16] new machineState: Cool Down -> PID Normal
[00:02:16] new PID-Values: P=55.0  I=0.8  D=715.0
[00:02:16] HW Brew - Voltage Sensor - Start
[00:02:16] new machineState: PID Normal -> Brew

Is this something weird on my machine or do you also see this?
The steam state should not be reached

kjyv, 2023-10-01:
Oh, well, I don't have my steam switch modded so the pin 35 is open and probably jumps around. I guess we need a new userConfig flag that says use the steam switch or not to prevent this

LoQue90, 2023-10-01:
with the new pcb the steam switch is always configured. when the external pulldown is on the pcb, steamswitch is always low.
But on a test board, i have the same log as you :)

gonna flash you PR directly to check the coldstart is doing:)

PR #376 (2023-12-05): Fix brew timer during coldstart
Fixes #366 by not running brew detection while heating up
closed 2023-12-05 by LoQue90