rancilio-pid/clevercoffee #370: ESP32 - Offline-Modus Probleme mit Wifi / WLAN

einencool, 2023-08-14:
Hallo,

ich habe einen ESP32 mit der Version 3.1.3 ausgestattet. 
Leider habe ich durch den inneren Verbau in meiner kleinen ESE-Pad Maschine ziemliche Probleme ein WLAN Signal zu erhalten.
eigentlich auch kein Problem, wenn denn der ESP direkt dann im Offline Modus starten würde.

Leider klappt das nach dem frischen Flashen nur ein oder zwei Mal und danach ist das Display beim Start die ganze Zeit nur am Blinken und fängt entsprechend nicht an hochzuheizen.

Habe schon in mehreren Issues nachgesehen und den ESP auch schon mehrfach gelöscht, neues Filesystem geflasht, dementsprechend neue Software, auch mal ohne und dann mit Display usw. aber trotzdem ändert sich nichts.

Wenn ich das Gerät auf den reinen Offline Modus flashe, läuft zwar alles, ich habe dann aber keinerlei Möglichkeit mehr, die Temperatur zu ändern.

Habe in der userConfig.h folgende Punkte mehrfach geändert und ausprobiert:
#define MAXWIFIRECONNECTS 0        // maximum number of reconnection attempts, use -1 to deactivate
#define WIFICONNECTIONDELAY 10000  // delay between reconnects in ms

Zusätzlich in der rancillio-pid.cpp (Zeilen 1590-1594):
    wm.setConfigPortalTimeout(10); // sec Timeout for Portal
    wm.setConnectTimeout(10); // Try 10 sec to connect to WLAN, 5 sec too short!
    wm.setBreakAfterConfig(true);
    wm.setConnectRetries(1);

Leider immer das Selbe Verhalten.
Gibt es da eine Möglichkeit, dass dies vernünftig funktioniert, oder was ich noch ändern muss?

Achso, kann man in die userConfig evtl. auch die WLAN Daten direkt eintragen?
Und kann man die PID Daten, die man in der rancillio-pid.h an seine Maschine angepasst hat, auch nach einem Werksreset Flash direkt aktivieren (PID-Haken, der sonst auf der Web Oberfläche gesetzt werden soll) ?
Dann könnte ich beim Flash im Offline Modus wenigstens die Maschine nutzen, weil ja auch so sonst der Haken in der Config fehlt...

Selbst mit einem WLAN Repeater direkt neben der Espresso Maschine bekomme ich es nicht zuverlässig hin.
Vielleicht hat der ESP auch eine WLAN Macke, aber für mich wäre es hauptsächlich, dass die Maschine IMMER hochheizt, auch wenn eben kein WLAN da ist. (Ein 97°C Thermostat ist zur Sicherheit noch in der Heizschleife verbaut, sodass nichts überhitzen kann, und die WLAN Steckdose, an der die Maschine hängt, schaltet sich nach 15min automatisch ab)

Falls ich noch weitere Infos liefern soll, oder etwas ausprobieren kann, würde ich mich über eine Rückmeldung sehr freuen.
Gruß
Christian

kjyv, 2023-09-09:
Ich hatte auch lange ähnliche Empfangsprobleme (je nachdem was bei dir im log steht und ob das wirklich passiert wenn der ESP in der Maschine ist), am Ende half eine Wechsel auf eine fritzbox (statt diesem Vodafone-Router). An sich sollte der ESP auch im Metallgehäuse gut funktionieren, wenn auch mit eingeschränkter Reichweite.
Aber hauptsächlich gehts ja hier darum, dass der Offlinemodus nicht anspringt.
WLAN-Credentials direkt eintragen geht leider momentan nicht in der userConfig

gang-himself, 2023-09-28:
Ich habe das gleiche Problem, allerdings mit dem ESP8266 auf 3.1.2
Solange WLAN existiert ist alles fein. Aber ohne WLAN versucht er nicht nur 5 Mal das WLAN zu finden (wie in der userconfig angegeben) sondern unendlich oft, der Offline-Modus springt also nicht an...

LG,
Wolfgang

einencool, 2023-09-29:
Schade zu lesen, also ist es aber wenigstens kein Problem, was spezifisch bei mir auftritt, sondern auch von anderen Usern beobachtet wird.

Hab allerdings gehofft, dass es einfacher wäre, den Fehler zu eliminieren, so ist es schon schade, dass man leider keine Möglichkeit hat, die Einstellungen der Maschine mehr zu ändern, denn ansonsten ist die Maschine nur unzuverlässig beim Hochfahren..

LoQue90, 2023-09-30:
> Ich habe das gleiche Problem, allerdings mit dem ESP8266 auf 3.1.2 Solange WLAN existiert ist alles fein. Aber ohne WLAN versucht er nicht nur 5 Mal das WLAN zu finden (wie in der userconfig angegeben) sondern unendlich oft, der Offline-Modus springt also nicht an...
> 
> LG, Wolfgang

okay, das ist ja wirklich ein bug dem man nachgehen muss.

Grundsätzlich wollt ihr aber auch ein bisschen etwas was sich gegenseitig ausschließt.
Die Maschine soll per WLAN erreichbar sein, aber eigentlich keine Zeit verbrauchen für den Verbindungsaufbau. Dazu ist die Lösung eigentlich nur guter WLAN Empfang.

kjyv, 2023-09-30:
Das Problem ist wahrscheinlich in diesem Code, der checkt ob setupDone nicht true ist. Direkt nach dem boot wird also kein Offline mode mehr aktiviert. Weiß aber nicht warum der check da so gemacht wurde und obs noch Gründe für gibt.
```
 if (wifiReconnects >= maxWifiReconnects && !setupDone) {  // no wifi connection after boot, initiate offline mode
        // (only directly after boot)
        initOfflineMode();
    }
```

kjyv, 2023-09-30:
Würds eher so vorschlagen:
```
    if (wifiReconnects >= maxWifiReconnects && WiFi.status() != WL_CONNECTED) { 
        // no wifi connection after trying connection, initiate offline mode
        initOfflineMode();
    } else 
        wifiReconnects = 0;
    }
```

LoQue90, 2023-10-01:
@einencool @gang-himself würdet ihr schauen ob der PR von kiyv euer Problem behebt? :)

gang-himself, 2023-10-01:
Gerne, weiß aber noch nicht, wann ich dazu komme. Meld mich dann wieder

einencool, 2023-10-01:
Da ich im Umgang mit Github leider nicht so erfahren bin, welche Version muss ich dann installieren? 
Ich hoffe, dass ich morgen im Laufe des Tages mal dazu kommen werde, oder am Feiertag :-)

gang-himself, 2023-10-01:
Du kannst in Visual Studio Code deine bisherige Version nehmen und in der Datei rancilio-pid.cpp den zu ersetzenden Code (der erste Post) durch den neuen ersetzen. 
Dann wieder alles compilieren und auf den ESP hochladen.

LG,
Wolfgang

gang-himself, 2023-10-01:
Scheint leider nichts zu verändern. 
Nur um sicher zu gehen, meine "Anleitung" im letzten Post ist richtig, oder?

gang-himself, 2023-10-01:
Bei mir hilft der PR wie gesagt nix. 
Jetzt versteh ich nicht, wieso das Thema als erledigt geschlossen wird.

LoQue90, 2023-10-01:
sprich er geht nach 5 mal verbindungsversuch bei dir nicht offline?
der richtige weg den PR zu testen, wäre dieses lokal zu clonen und auf die maschine zu spielen.

gang-himself, 2023-10-01:
Das stimmt leider. Er versucht permanent zu verbinden. 

Ja, da es jetzt eingepflegt ist, kann ich es lokal klonen. Aber hätte ich die Möglichkeit vorher auch schon gehabt, oder war mein Ansatz auch richtig? Also den Code in der lokalen Datei zu ersetzen, die Datei speichern und über Build und Upload auf die Maschine zu spielen?

gang-himself, 2023-10-01:
> Grundsätzlich wollt ihr aber auch ein bisschen etwas was sich gegenseitig ausschließt. Die Maschine soll per WLAN erreichbar sein, aber eigentlich keine Zeit verbrauchen für den Verbindungsaufbau. Dazu ist die Lösung eigentlich nur guter WLAN Empfang.

Nicht wirklich. Zeit für Verbindungssufbau ist in Ordnung. aber eine „tote“ Maschine, nur weil das WLAN nicht geht, finde ich nicht in Ordnung…

fiendie, 2023-10-01:
Ich weiß nicht, was genau du gemacht hast, aber die rancilio-pid.cpp gibt es z.b. nicht mehr. Es klingt so, als hättest du einen älteren Stand der Firmware.  
Wenn du einen PR testen willst, solltest du dir den Code nicht selbst zusammenkopieren sondern einfach das Repository klonen und den PR lokal auschecken.

kjyv, 2023-10-01:
Also zumindest so viel: ich hab das Problem vorher reproduzieren können und nach der Änderung sehe ich den Offline-Modus wenn meine Maschine das Wlan nicht mehr findet was konfiguriert ist (da der Router aus war). Das war doch das Szenario um das es ging?
Ein Unterschied kann sein, dass das Verhalten direkt beim Start und wenn die Verbindung irgendwann später abbricht separat ist. Ich hab mir das nur für den Start angeguckt.

kjyv, 2023-10-01:
Aber ja, ich würd auch sagen nimm zum Testen lieber den aktuellen Entwicklungsstand als was zusammenzukopieren, dann wissen wir immer genau wovon wir ausgehen können. Da es ja jetzt eh gemerged ist kannst du einfach auf den master branch gehen und das bauen.

einencool, 2023-10-02:
> Also zumindest so viel: ich hab das Problem vorher reproduzieren können und nach der Änderung sehe ich den Offline-Modus wenn meine Maschine das Wlan nicht mehr findet was konfiguriert ist (da der [...]

Guten Morgen,
Ich bin zwar noch nicht zum testen gekommen, aber das gleiche Phänomen konnte ich auch mit der originalen alten Version beobachten.

Es hat für ein paar Male direkt nach dem flashen funktioniert und danach dann nicht mehr.

Das war für mich ja das "ärgerliche", weil ich die Maschine dafür immer wieder öffnen musste um an das Board zu kommen.
Es hatte beim Testen ein paar mal funktioniert, auch der offline Modus ging an, aber am nächsten Tag fing das Spiel wieder von vorne an...

Jetzt habe ich intern ein Kabel verbaut und auf der Rückseite nach draußen geführt.

Ich hoffe, dass ich heute oder morgen zum testen komme.

Vielen Dank, dass jetzt nach der Zeit doch so viel Bewegung in das Thema gekommen ist.

gang-himself, 2023-10-02:
> Aber ja, ich würd auch sagen nimm zum Testen lieber den aktuellen Entwicklungsstand als was zusammenzukopieren, dann wissen wir immer genau wovon wir ausgehen können. Da es ja jetzt eh gemerged ist kannst du einfach den master branch nehmen und das bauen.

Entschuldigung, mit Github bin ich auch nicht so bewandert, ich hatte direkt übersehen, dass kjyv direkt eine Version erstellt hatte.
Soweit ich das sehen kann, ist der Fix aber nur im Masterbranch für den ESP32 gemerged, den ich mit dem ESP8266 ja nicht verwenden kann. Gibt es auch einen PR für die alte ESP8266-Version?

kjyv, 2023-10-02:
Bisher hast du von ESP32 geschrieben, aber im Prinzip sollte die Änderung auch mit der Änderung in der am besten jetzt aktuellen 3.2.0 funktionieren. Wenn du das mit dem Kabel draufgespielt hast und das Problem weiterhin besteht, wäre es gut mal die Log-Ausgabe (vom Monitor-Modus von PlatformIO) hier zu sehen.

gang-himself, 2023-10-02:
Mach ich heute abend, Danke. 
Kann ich auch OTA auf die 3.2.0 updaten oder muss das über Kabel passieren? Dafür müsste ich dann noch ein paar Tage warten…

LoQue90, 2023-10-02:
OTA geht auch

gang-himself, 2023-10-02:
So, 3.2.0 ist drauf, den Code habe ich geändert, leider bleibt es bei dem Fehlverhalten.
Am Donnerstag habe ich Zeit, mal die Maschine aufzumachen und über USB zu schauen, was die Logs hergeben...

gang-himself, 2023-10-05:
Nach drei Versuchen, zum WLAN zu verbinden öffnet der ESP das Config-Portal, statt in den Offline-Modus zu gehen.
Dann gibt es einen Kernel-Panic und der ESP rebootet...

```
*wm:[2] Connection result: WL_NO_SSID_AVAIL
*wm:[1] AutoConnect: FAILED 
*wm:[2] Starting Config Portal 
*wm:[2] AccessPoint set password is VALID 
*wm:[2] Disabling STA 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  silvia
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] WiFi Scan ASYNC started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
*wm:[2] Portal Timeout In 60 seconds
*wm:[2] Portal Timeout In 58 seconds

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:133 __yield

>>>stack>>>
3ffff9e0:  41455400 504b5f4d 80efef00 61657400  
3ffff9f0:  704b206d 80efef00 feefef01 3fff4200  
3ffffa00:  003d003f 80efeffe 00000002 00000000  
3ffffa10:  3fff4294 feefeffe 4020a344 40232f94  
3ffffa20:  00000000 000001f4 3ffe85c8 3fff4200  
3ffffa30:  000f000f 80efeffe 3fff4200 0015001f  
3ffffa40:  80efeffe feefef01 3fff4200 0034003f  
3ffffa50:  80efeffe 00000002 00000001 3fff4324  
3ffffa60:  feefeffe 4020a37c 40232f98 00000064  
3ffffa70:  0000008c 3ffe85d0 504d4500 feefef00  
3ffffa80:  80efeffe 3fff4300 000b000f 80efeffe  
3ffffa90:  feefef00 feefef00 feefeffe 80efeffe  
3ffffaa0:  00000002 00000000 3fff4344 feefeffe  
3ffffab0:  4020a3b4 40232f9c 00000000 000000c8  
3ffffac0:  3fff1530 3fff4300 000e000f 80efeffe  
3ffffad0:  3fff4300 000f000f 80efeffe feefef01  
3ffffae0:  3fff4300 003b003f 80efeffe 00000002  
3ffffaf0:  00000001 3fff43c4 feefeffe 4020a3ec  
3ffffb00:  40232fa0 00000014 0000006e 3ffe85d8  
3ffffb10:  3fff4300 0010001f 80efeffe 3fff4300  
3ffffb20:  000c000f 80efeffe feefef01 3fff4400  
3ffffb30:  00e000ef 80efeffe 00000002 00000001  
3ffffb40:  3fff4504 feefeffe 4020a424 40232fa4  
3ffffb50:  00000000 00000014 3fff18f0 57455200  
3ffffb60:  4d49545f 80ef0045 3fff4500 000d000f  
3ffffb70:  80efeffe feefef01 3fff4500 0019001f  
3ffffb80:  80efeffe 00000002 00000001 3fff454c  
3ffffb90:  feefeffe 4020a458 40232fa8 00000001  
3ffffba0:  000000b4 3ffe8590 3fff4500 0015001f  
3ffffbb0:  80efeffe 3fff4500 001a001f 80efeffe  
3ffffbc0:  feefef00 feefef00 feefeffe 80efeffe  
3ffffbd0:  00000002 00000001 3fff45a4 feefeffe  
3ffffbe0:  4020a490 40232fac 00000000 0000003c  
3ffffbf0:  3ffe8580 3fff4500 0010001f 80efeffe  
3ffffc00:  3fff4500 0014001f 80efeffe feefef00  
3ffffc10:  feefef00 feefeffe 80efeffe 00000002  
3ffffc20:  00000001 3fff45fc feefeffe 4020a4c8  
3ffffc30:  40232fb0 00000000 0000003c 3ffe8588  
3ffffc40:  3fff4600 0014001f 80efeffe 3fff4600  
3ffffc50:  0018001f 80efeffe feefef01 3fff4600  
3ffffc60:  0029002f 80efeffe 00000002 00000001  
3ffffc70:  3fff468c feefeffe 4020a500 40232fb4  
3ffffc80:  00000000 000001f4 3ffe8578 5f444900  
3ffffc90:  4f5f4442 80ef004e 3fff4600 000f000f  
3ffffca0:  80efeffe feefef01 3fff4600 0031003f  
[... 101 lines cut ...]
0x4020a4c8 in std::_Function_base::_Base_manager<setup::{lambda()#16}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#16}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40232fb0 in std::_Function_handler<bool (), setup::{lambda()#16}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a500 in std::_Function_base::_Base_manager<setup::{lambda()#17}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#17}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40232fb4 in std::_Function_handler<bool (), setup::{lambda()#17}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a538 in std::_Function_base::_Base_manager<setup::{lambda()#18}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#18}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40232fb8 in std::_Function_handler<bool (), setup::{lambda()#18}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a570 in std::_Function_base::_Base_manager<setup::{lambda()#19}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#19}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40209e58 in std::_Function_handler<bool (), setup::{lambda()#19}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a5a8 in std::_Function_base::_Base_manager<setup::{lambda()#20}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#20}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40209e68 in std::_Function_handler<bool (), setup::{lambda()#20}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a5e0 in std::_Function_base::_Base_manager<setup::{lambda()#21}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#21}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40209e78 in std::_Function_handler<bool (), setup::{lambda()#21}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a618 in std::_Function_base::_Base_manager<setup::{lambda()#22}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#22}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40209e88 in std::_Function_handler<bool (), setup::{lambda()#22}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
[... cut ...]
0x40232fc4 in std::_Function_handler<bool (), setup::{lambda()#25}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4020a6f4 in std::_Function_base::_Base_manager<setup::{lambda()#26}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#26}> const&, std::_Manager_operation) at rancilio-pid.cpp:?
0x40232fc8 in std::_Function_handler<bool (), setup::{lambda()#26}>::_M_invoke(std::_Any_data const&) at rancilio-pid.cpp:?
0x4022979c in loop_wrapper() at core_esp8266_main.cpp:?

`

gang-himself, 2023-10-05:
Habe jetzt mal noch einen Flash-Erase durchgeführt und anschließend alles neu draufgeschrieben.
Leider immer noch das gleiche Verhalten. 
Auch das Abschalten des OLED-Displays (wie es bei der Ersteinrichtung nötig ist) verändert nichts an dem Verhalten.
Sobald das Config-Portal gestartet wird, rebootet der ESP.

Hier nochmal die Meldungen wie sie jetzt aussehen:

```*wm:[2] Connection result: WL_NO_SSID_AVAIL
*wm:[1] AutoConnect: FAILED 
*wm:[2] Starting Config Portal 
*wm:[2] AccessPoint set password is VALID 
*wm:[2] Disabling STA 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  silvia

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
*wm:[1] AutoConnect 
*wm:[2] Setting Hostnames:  silvia
*wm:[2] Setting WiFi hostname 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connect Wifi, ATTEMPT # 1 of 3`

gang-himself, 2023-10-05:
Ich glaube den Fehler gefunden zu haben:

Nach einem kompletten Flash-Erase und einem Neuaufsetzen musste ich ja das OLED-Display abschalten, um die WIFI-Ersteinrichtung machen zu können.
Hier liegt das Problem: Bevor der ESP in den Offline-Modus geht, startet er ja das Config-Portal. Und bei angeschaltetem OLED-Display stürzt dabei der ESP ab und rebootet.
Sobald ich das OLED-Display deaktiviere, geht er auch brav in den Offline-Modus...

kjyv, 2023-10-05:
Ja, mit ESP8266 ist das natürlich ein Problem. Allerdings wollen wir ja auch dass es möglich ist, das Wlan neu zu konfigurieren, wenn das eingestellte nicht erreichbar ist. Es gab die Idee, den WifiManager für ESP8266 komplett zu entfernen, vielleicht wäre das dann der richtige fix dafür. Für ESP32 gibt es das Problem aber nicht.

gang-himself, 2023-10-05:
> Allerdings wollen wir ja auch dass es möglich ist, das Wlan neu zu konfigurieren, wenn das eingestellte nicht erreichbar ist. 

Das wäre schön, aber geht momentan ja auch nicht, solange ein OLED konfiguriert ist. 

> Es gab die Idee, den WifiManager für ESP8266 komplett zu entfernen, vielleicht wäre das dann der richtige fix dafür. 

Fände ich richtig. Man muss ja sowieso mit Kabel an das Board, wenn das eingestellte WLAN nicht erreichbar ist, um entweder das Display zu deaktivieren oder eben gleich das WLAN zu setzen, falls das dann über die config möglich ist

gang-himself, 2023-10-06:
Vielleicht kann ja der Start des config-Portals abhängig von OLED-Display 0 machen. 
Dann startet es nur, wenn es auch funktioniert und der ESP schmiert nicht ab

LoQue90, 2023-10-06:
Das geht nicht, allein beide eingebunden libs führen zum crash

LoQue90, 2023-12-05:
der WM ist für den esp8266 wieder rausgefolgen.
https://github.com/rancilio-pid/clevercoffee/pull/405
das sollte zeitnah ein neues release zu kommen.
Gerne Feedback dazu.

gang-himself, 2023-12-06:
Juhuu, funktioniert!
Danke dafür.

Kann aus meiner Sicht geschlossen werden.

Einziges Problem ist jetzt noch, dass die Monitor-Funktion in PlatformIO mit einem Errno 61 - Connection refused abbricht...
Hab ich da was übersehen?

LG

LoQue90, 2023-12-06:
> Juhuu, funktioniert! Danke dafür.
> 
> Kann aus meiner Sicht geschlossen werden.
> 
> Einziges Problem ist jetzt noch, dass die Monitor-Funktion in PlatformIO mit einem Errno 61 - Connection refused abbricht... Hab ich da was übersehen?
> 
> LG

hast du die passwörter überprüft? das in der userconfig und das in der platformini?

gang-himself, 2023-12-06:
Ist jeweils das Standardpasswort otapass

gang-himself, 2023-12-07:
Ah, ok. Damit kann ich leben :-)
Wär aber für die Doku bestimmt interessant, bzw. könnte man die Funktion aus dem PlatformIO-Projekt rausschmeißen...

Danke nochmal!

LoQue90, 2023-12-07:
wird dann bei release entsprechend dokumentiert.
Aus Platformio würde ich es nicht rauswerfen, frühere versionen können es ja.

PR #377 (2023-10-01): fix never entering offline mode
Should fix #370
closed 2023-12-08 by LoQue90