rancilio-pid/clevercoffee #372: clevercoffee.de is blank

lassem, 2023-09-24:
Website only renders a white page.
Tested on MacOS with Chrome and Safari and Chrome on a Pixel 7

fiendie, 2023-09-25:
Could you hook up the board via USB and copy-paste the output of the serial monitor?

chris-dot-exe, 2023-09-25:
> Could you hook up the board via USB and copy-paste the output of the serial monitor?

I'm pretty sure @lassem is taking about the website [https://clevercoffee.de/](https://clevercoffee.de/)  not the website served from the ESP cause I also only get a white page since yesterday

[image]

fiendie, 2023-09-25:
My bad. @btotherunner is running the website and he's looking into it.

murmeltier08, 2023-09-30:
website is running right now

closed 2023-09-30 by murmeltier08