rancilio-pid/clevercoffee #451: Pump starts when tank is empty and brew was manually canceled

maunsen, 2024-02-02:
I am running clvercoffe on an ESP32 and have the complete rebuild with Trigger-Button for brewing and waterlevel sensor XKC-Y25-PNP.
Version is 4.0.0

Situation:
Start a brew.
During this brew the waterlevel has to go below the level that the senor gets triggered.
When I then stop the brew manually the machine shows the empty tank symbol.
When I then refill the tank, it triggers a new brew instantly.

Expected behaviour:
When I refill the tank, the empty tank Icon vanishes, and the machine goes back to its normal idle state.

genkigenki, 2024-02-03:
Version is latest [b6c3066](https://github.com/genkigenki/clevercoffee/commit/b6c3066af37699e203165b8c56131013fa2d1727)
Trigger Brew switch
Preinfusion

Situation:
Start a brew. 
Preinfusion finished, brew starts
During brew waterlevel made low - low-water icon is shown on display
Stop brew manually - machine shows the empty tank symbol.
Refill tank, no brew triggered.

So i cant reproduce the bug, at least with brew-trigger, latest SW, preinfusion active.

LoQue90, 2024-02-11:
should be fixed with #455

LoQue90, 2024-02-11:
PR merged into master. @maunsen can you test ist? didn´t have the problem on my mashine

closed 2024-02-11 by LoQue90