rancilio-pid/clevercoffee #523: Can't access website with 3.3.0 on ESP8266

smitterer, 2024-12-23:
Hi!

Unfortunately I can't open the website to set the parameters on 3.3.0 using a ESP8266.
According the my DHCP server it is connected and has a IP assigned.

This also leads to the problem that I can't activate the PID.

Any idea?

Thanks

LoQue90, 2024-12-23:
Hi,

what is the error you get when you try to reach the esp on your browser?
Which browser do you use?

smitterer, 2024-12-23:
@LoQue90 it keeps loading and then says ERR_CONNECTION_TIMED_OUT
Tried it with Chrome and Edge

LoQue90, 2024-12-23:
could you hook ab usb serial in vsc code and use the monitor to see what the esp is doing when you try to reach the website?

smitterer, 2024-12-23:
Nothing about the website.

here is the ouput:
```
--- Terminal on COM8 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp8266_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[00:00:06] WiFi connected - IP = 192.168.0.22
[00:00:06] MAC-ADDRESS: 132131421762144
[00:00:06] getNumber(): addr=154 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=14 value=45.0
[00:00:06] getNumber(): addr=174 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=15 value=130.0
[00:00:06] getNumber(): addr=0 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=0 value=130.0
[00:00:06] getNumber(): addr=10 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=1 value=130.0
[00:00:06] getNumber(): addr=20 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=2 value=0.5
[00:00:06] getNumber(): addr=28 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=22 value=50.0
[00:00:06] getNumber(): addr=104 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=9 value=50.0
[00:00:06] getNumber(): addr=114 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=10 value=0.0
[00:00:06] getNumber(): addr=124 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=11 value=20.0
[00:00:06] getNumber(): addr=37 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=3 value=95.0
[00:00:06] getNumber(): addr=45 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=23 value=0.0
[00:00:06] getNumber(): addr=54 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=4 value=25.0
[00:00:06] getNumber(): addr=134 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=12 value=25.0
[00:00:06] getNumber(): addr=144 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=13 value=120.0
[00:00:06] getNumber(): addr=64 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=5 value=2.0
[00:00:06] getNumber(): addr=74 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=6 value=5.0
[00:00:06] getNumber(): addr=236 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=20 value=30.0
[00:00:06] getNumber(): addr=18 size=1/1
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=19 value=0
[00:00:06] getNumber(): addr=244 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=21 value=150.0
[00:00:06] getNumber(): addr=252 size=8/8
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=26 value=120.0
[00:00:06] getNumber(): addr=153 size=1/1
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=24 value=0
[00:00:06] getNumber(): addr=103 size=1/1
[00:00:06] getNumber(): storage empty -> returning default
[00:00:06] storageGet(): item=25 value=0
[00:00:06] Server started at 192.168.0.22
[00:00:07] new machineState: Init -> PID Offline
[00:00:08] *** WARNING: temperature sensor reading: consec_errors = 1, temp_current = 221.0, temp_prev = 221.0
[00:00:09] *** WARNING: temperature sensor reading: consec_errors = 2, temp_current = 221.0, temp_prev = 221.0
[00:00:10] *** WARNING: temperature sensor reading: consec_errors = 3, temp_current = 221.0, temp_prev = 221.0
[00:00:11] *** WARNING: temperature sensor reading: consec_errors = 4, temp_current = 221.0, temp_prev = 221.0
[00:00:12] *** WARNING: temperature sensor reading: consec_errors = 5, temp_current = 221.0, temp_prev = 221.0
[00:00:13] *** WARNING: temperature sensor reading: consec_errors = 6, temp_current = 221.0, temp_prev = 221.0
[00:00:14] *** WARNING: temperature sensor reading: consec_errors = 7, temp_current = 221.0, temp_prev = 221.0
[00:00:15] *** WARNING: temperature sensor reading: consec_errors = 8, temp_current = 221.0, temp_prev = 221.0
[00:00:16] *** WARNING: temperature sensor reading: consec_errors = 9, temp_current = 221.0, temp_prev = 221.0
[00:00:17] *** WARNING: temperature sensor reading: consec_errors = 10, temp_current = 221.0, temp_prev = 221.0
[00:00:17] *** ERROR: temperature sensor malfunction: temp_current = 221.0
[00:00:17] new machineState: PID Offline -> Sensor Error
[00:00:17] Attempting MQTT reconnection: 1
[00:00:17] Subscribe to MQTT Topics
```

LoQue90, 2024-12-23:
do you have a tsic connected?

LoQue90, 2024-12-23:
there is nothing else in the log when you try to open the website?

LoQue90, 2024-12-23:
esp is answering to ping?

smitterer, 2024-12-23:
Well this is interesing.
Pinging it results in:
```
Pinging 192.168.0.22 with 32 bytes of data:
Reply from 192.168.0.7: Destination host unreachable.
Reply from 192.168.0.7: Destination host unreachable.
```

But in Winbox (Mikrotik Router) I see the gaggia under 192.168.0.22

LoQue90, 2024-12-23:
diffrent vlans?
mixed 2,4/5Ghz ssid?
auto moving clients to 5ghz?

smitterer, 2024-12-23:
SSID for 2.4 and 5 are the same.

MQTT is working.
Therefore I don't think it is a network problem.

LoQue90, 2024-12-23:
and you did the "upload filesystem image" ?

smitterer, 2024-12-23:
I did, yes.

Maybe it is the same issue as here?: https://github.com/tzapu/WiFiManager/issues/1579#issuecomment-1587107763
Where would I set the different port?

LoQue90, 2024-12-23:
3.3.0 for esp8266 dosn´t use wifimanager :)

smitterer, 2024-12-23:
I will check the router more in detail to see if the problem relies there.

Because I can ping the esp directly from the mikrotik, but not from the computer.
MQTT is still working.

smitterer, 2024-12-24:
@LoQue90 so this is very strange. I can now access the website (for whatever reason) but I can't enable the PID controller. Clicking the corresponding checkbox and then saving does reset it to off.
Other values like PID TN can be set and saved.

LoQue90, 2024-12-24:
You did enabled the powerswitch in the userconfig?
That is the reason why you can't turn it on on the website.

smitterer, 2024-12-24:
Thanks, this allowed me to set it!

closed 2024-12-24 by smitterer