rancilio-pid/clevercoffee #525: TRIGGERTYPE (not) starting pump

smitterer, 2024-12-24:
Hi!

When I set TRIGGERTYPE to low, activating the pump via the switch does not start the pump while setting it to high, immediately starts it when I turn on my Gaggia.

Could you eventually check my config for any errors?
I am using 3.3.0 with an ESP8266.

```
/**
 * @file    userConfig.h
 * @brief   Values must be configured by the user
 * @version 3.3.0 Master
 *
 */
#ifndef _userConfig_H
#define _userConfig_H

// firmware version (must match with definitions in the main source file)
#define USR_FW_VERSION    3
#define USR_FW_SUBVERSION 3
#define USR_FW_HOTFIX     0
#define USR_FW_BRANCH     "ESP8222-MASTER"

// List of supported machines
enum MACHINE {
    RancilioSilvia,   // MACHINEID 0
    RancilioSilviaE,  // MACHINEID 1
    Gaggia,           // MACHINEID 2
    QuickMill         // MACHINEID 3
};

/**
 * Preconfiguration
 */

// Machine
#define MACHINEID 2                // see above list of supported machines

// Display
#define OLED_DISPLAY 2             // 0 = deactivated, 1 = SH1106 (e.g. 1.3 "128x64), 2 = SSD1306 (e.g. 0.96" 128x64), 3 = SH1106_126x64_SPI
#define OLED_I2C 0x3C              // I2C address for OLED, 0x3C by default
#define DISPLAYTEMPLATE 1          // 1 = Standard display template, 2 = Minimal template, 3 = only temperature, 4 = scale template, 20 = vertical display (see git Handbook for further information)
#define DISPLAYROTATE U8G2_R0      // rotate display clockwise: U8G2_R0 = no rotation; U8G2_R1 = 90°; U8G2_R2 = 180°; U8G2_R3 = 270°
#define SCREEN_WIDTH 128           // OLED display width, in pixels
#define SCREEN_HEIGHT 64           // OLED display height, in pixels
#define SHOTTIMER 1                // 0 = deactivated, 1 = activated 2 = with scale
#define HEATINGLOGO 2              // 0 = deactivated, 1 = Rancilio, 2 = Gaggia
#define OFFLINEGLOGO 1             // 0 = deactivated, 1 = activated
#define BREWSWITCHDELAY 3000       // time in ms that the brew switch will be delayed (shot timer will show that much longer after switching off)
#define VERBOSE 0                  // 1 = Show verbose output (serial connection), 0 = show less

#define LANGUAGE 0                 // LANGUAGE = 0 (DE), LANGUAGE = 1 (EN), LANGUAGE = 2 (ES)

// Connectivity
#define CONNECTMODE 1              // 0 = offline 1 = WIFI-MODE
#define HOSTNAME "gaggia"
#define WIFI_SSID "mywifi"       // SSID of your WiFi network
#define WIFI_PASS "mypw"   // WPA key to your WiFi network
#define MAXWIFIRECONNECTS 5        // maximum number of reconnection attempts, use -1 to deactivate
#define WIFICONNECTIONDELAY 10000  // delay between reconnects in ms

// PID & Hardware
#define ONLYPID 0                  // 1 = Only PID, 0 = PID and preinfusion
#define ONLYPIDSCALE 0             // 0 = off , 1 = OnlyPID with Scale
#define BREWMODE 1                 // 1 = Brew by time (with preinfusion); 2 = Brew by weight (from scale)
#define BREWDETECTION 2            // 0 = off, 1 = Software (Onlypid 1), 2 = Hardware (Onlypid 0), 3 = Sensor/Hardware for Only PID
#define BREWSWITCHTYPE 1           // 0 = no switch connected, 1 = normal switch, 2 = trigger switch
#define POWERSWITCHTYPE 0          // 0 = no switch connected, 1 = normal switch, 2 = trigger switch
#define TRIGGERTYPE HIGH           // LOW = low trigger, HIGH = high trigger relay for pump & valve 
#define VOLTAGESENSORTYPE HIGH     // BREWDETECTION 3 configuration
#define PINMODEVOLTAGESENSOR INPUT // Mode INPUT_PULLUP, INPUT or INPUT_PULLDOWN_16 (Only Pin 16)
#define PRESSURESENSOR 0           // 1 = pressure sensor connected to A0; PINBREWSWITCH must be set to the connected input!
#define TEMPLED 0                  // set led pin high when brew or steam set point is within range

// TOF sensor for water level
#define TOF 0                      // 0 = no TOF sensor connected; 1 = water level by TOF sensor
#define TOF_I2C 0x29               // I2C address of TOF sensor; 0x29 by default
#define CALIBRATION_MODE 0         // 1 = enable to obtain water level calibration values; 0 = disable for normal PID operation; can also be done in Blynk
#define WATER_FULL 102             // value for full water tank (=100%) obtained in calibration procedure (in mm); can also be set in Blynk
#define WATER_EMPTY 205            // value for empty water tank (=0%) obtained in calibration procedure (in mm); can also be set in Blynk

// E-Trigger
#define ETRIGGER 0                 // 0 = no trigger (for Rancilio except Rancilio E), 1 = trigger for CPU of Rancilio E
#define ETRIGGERTIME 600           // seconds, time between trigger signal
#define TRIGGERRELAYTYPE HIGH      // LOW = low trigger, HIGH = high trigger relay for E-Trigger

// Brew Scale
#define SCALE_SAMPLES 2                     // Load cell sample rate
#define SCALE_CALIBRATION_FACTOR 3195.83    // Raw data is divided by this value to convert to readable data

/* Pressure sensor
 *
 * measure and verify "offset" value, should be 10% of ADC bit reading @supply volate (3.3V)
 * same goes for "fullScale", should be 90%
 */
#define OFFSET      102            // 10% of ADC input @3.3V supply = 102
#define FULLSCALE   922            // 90% of ADC input @3.3V supply = 922
#define MAXPRESSURE 200

// PlatformIO OTA
#define OTA true                   // true = OTA activated, false = OTA deactivated
#define OTAHOST "otahost"           // Name to be shown in Arduino IDE/PlatformIO port
#define OTAPASS "otapw"          // Password for OTA updates

// MQTT
#define MQTT 1                             // 1 = MQTT enabled, 0 = MQTT disabled
#define MQTT_USERNAME ""
#define MQTT_PASSWORD ""
#define MQTT_TOPIC_PREFIX "kitchen/gaggia."  // topic will be "<MQTT_TOPIC_PREFIX><HOSTNAME>/<READING>"
#define MQTT_SERVER_IP "mymqttserver"   // IP-Address of locally installed mqtt server
#define MQTT_SERVER_PORT 1883

// INFLUXDB
#define INFLUXDB 0                 // 1 = INFLUX enabled, 0 = INFLUX disabled
#define INFLUXDB_URL ""            // InfluxDB server address
#define INFLUXDB_AUTH_TYPE 1       // 1 = API Token , 2 = User/Pass
#define INFLUXDB_API_TOKEN ""
#define INFLUXDB_ORG_NAME ""
#define INFLUXDB_USER ""
#define INFLUXDB_PASSWORD ""
#define INFLUXDB_DB_NAME "coffee"  // InfluxDB bucket name
#define INFLUXDB_INTERVAL 5000     // Send interval in milliseconds

// PID Parameters (not yet in Web interface)
#define EMA_FACTOR 0.6             // Smoothing of input that is used for Tv (derivative component of PID). Smaller means less smoothing but also less delay, 0 means no filtering
#define BREWPID_DELAY 10           // delay until enabling PID controller during brew (no heating during this time)
#define TEMPSENSOR 2               // Temp sensor type: 1 = DS18B20, 2 = TSIC306

// Backflush values
#define FILLTIME 3000              // time in ms the pump is running
#define FLUSHTIME 6000             // time in ms the 3-way valve is open -> backflush
#define MAXFLUSHCYCLES 5           // number of cycles the backflush should run, 0 = disabled

#endif
```

Thank you very much!

LoQue90, 2024-12-24:
Config looks fine.
Setting TRIGGERTYPE to High resulting in turning on the pump directly while turn on the machine is a expected behavior with a high trigger relay.

What does the log showing when setting to low trigger and starting a brew?

smitterer, 2024-12-25:
To be honest, I'm trouble connecting via OTA and always need to put out the ESP and connect it via USB.
Therefore, I can't test that.

I did several changes to platform.ini like
monitor_port = socket://gaggia:23
monitor_port = socket://gaggialocal:23
monitor_port = socket://theipaddress:23
upload_port = gaggia
upload_flags = --auth=otapass
upload_flags = --auth=MYACTUALPW

Based on my config, what would be the correct setting in platform.ini?

Thanks!

LoQue90, 2024-12-25:
3.3.0 has no ota monitor option.

smitterer, 2024-12-25:
Unfortunately no output at all when switching:
```
[00:00:06] new machineState: Init -> Cold Start
[00:00:06] new PID-Values: P=130.0  I=1.0  D=70.2
[00:00:16] Attempting MQTT reconnection: 1
[00:00:16] Subscribe to MQTT Topics
[00:01:55] temperature >= (BrewSetPoint-1), wait 10 sec before machineState BelowSetPoint
[00:02:05] 5 sec temperature >= (BrewSetPoint-1) finished, switch to state BelowSetPoint
[00:02:05] new machineState: Cold Start -> Set Point Negative
[00:02:05] new PID-Values: P=130.0  I=1.0  D=70.2
[00:02:27] new machineState: Set Point Negative -> PID Normal
[00:02:27] new PID-Values: P=130.0  I=1.0  D=70.2
```

I also tried bridging the to solder points to test if ther soldering is loose or so but also no log output.

LoQue90, 2024-12-25:
Then your brew switch doesn't work.
There should be a log that the brew startet when you hit the brew switch.

smitterer, 2024-12-25:
Pin 15?
The plan says it should be connected to A0/3V or am I wrong on that?

Thanks

LoQue90, 2024-12-25:
We did introduced a standard pimmapping with 3.x
Brewswitch is on pin 15.

smitterer, 2024-12-25:
Is there somewhere a new plan and are there any other changes?
Which of the pins on 15 should be connected to the switch?

LoQue90, 2024-12-25:
Other changes are writen in the different releases on github.
For esp8266 there is no new schematic.
Pimmapping in total could be found in the esp826nodemcu.h
Output of your switch should be connected to the "sig" terminal of pin 25

smitterer, 2024-12-25:
Thanks, it is working now!

closed 2024-12-25 by smitterer