rancilio-pid/clevercoffee #529: esp_littlefs: corrupted dir pair at {0x0, 0x1}

spezi83, 2025-02-14:
Hi, 
I tried to use a original espressif devkit ESP32DevKitC with ESP32-WROOM-32E
I can setup the wifi, but in the monitor I get the following Error:
```
E (10391) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1367:error: Corrupted dir pair at {0x0, 0x1}

E (10394) esp_littlefs: mount failed,  (-84)
E (10398) esp_littlefs: Failed to initialize LittleFS
[ 10419][E][LittleFS.cpp:98] begin(): Mounting LittleFS failed! Error: -1
[00:00:10]    INFO Server started at 192.168.41.148
[ 10448][I][ArduinoOTA.cpp:141] begin(): OTA server at: silvia.local:3232
[00:00:10] WARNING Temperature reading failed
[00:00:10]    INFO LittleFS: 2147483647% (used 0 bytes from 0 bytes)
```
I cannot browse the IP of the ESP32 afterwards. At the moment I have no clue what is wrong

lbsonix, 2025-02-15:
I'm facing the same issue

`./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
E (142024) esp_littlefs: mount failed,  (-84)
E (142028) esp_littlefs: Failed to initialize LittleFS
[142045][E][LittleFS.cpp:95] begin(): Mounting LittleFS failed! Error: -1
[142052][E][AsyncTCP.cpp:1268] begin(): bind error: -8`

lbsonix, 2025-02-15:
Had the same issue:

Erase the flash
Build Filesystem Image
Upload Filesystem Image
and then do the build + upload of the firmware.

remove the begin(true) before build + upload

Worked for me this way
check docs https://manual.rancilio-pid.de/de/software-part-I/programmcode.html

LoQue90, 2025-02-15:
Did you uploaded the file system image?

When you format the filesystem during boot, then of course the Webserver can't find any files.

spezi83, 2025-02-15:
> Had the same issue:
> 
> Erase the flash Build Filesystem Image Upload Filesystem Image and then do the build + upload of the firmware.
> 
> remove the begin(true) before build + upload
> 
> Worked for me this way check docs https://manual.rancilio-pid.de/de/software-part-I/programmcode.html

Thanks a lot. Seems like the build environment is a bit picky. Now it works. Issue can be close

closed 2025-02-15 by LoQue90