rancilio-pid/clevercoffee #535: Scale weight doesn't zero for brew after flush

scottcondie24, 2025-04-21:
Setup is momentary brew switch and scales. 

When holding the brew switch to flush it starts the timeBrewed but when released it doesn't reset timeBrewed to zero, so shottimerCounter doesn't reset to 10. The next time you start a brew it doesn't auto-tare. 

To fix it I added currBrewState = kWaitBrewOff inside the kBrewSwitchFlushOff but there may be a more elegant solution.

scottcondie24, 2025-04-21:
Thanks mate, it's possible it is solved with #512 as there are quite a few changes in how it's handled. I didn't see that one as I only filtered for open issues (not enhancements). I'll have a look at that branch and learn how to swap between them to test it

LoQue90, 2025-04-21:
I do reset timeBrewed in the beginning of a brew or flush.
It is in the idle states.
So in timeBrewed is always the last brew stored until a new one is started.
The reason is that I don't need a new var to store the last brew to show the time on the display after the brew is finished.
I did considered that this is maybe a problem for the scale, because I don't have one.

scottcondie24, 2025-04-21:
That makes a lot of sense, I did notice the master stored the timeBrewed in a new variable before resetting but this way you won't need it. 

As for the scales it may not trigger a reset if it counts up before checking the scales but I haven't looked that far yet, maybe there will be a better variable to trigger from

LoQue90, 2025-04-21:
I wanted to move away from trigger stuff based on vars.
I build the whole brew stuff new based on states.
So I would like to have the scale also be triggered based on states from the brew controll and not on timeBrewed gets 0.
Looks like I missed that part for the scale.
Looking forward to fix this together.

scottcondie24, 2025-04-21:
I like the new method, it should make it easier to modify.

Based on that it should be straightforward to modify the scales using currBrewState == kBrewRunning instead of timeBrewed == 0 if they are global vars. It would be handy to capture the final drips weight after the brew finishes but thats separate from this.

I will have to spend the time to make a proper branch with my changes, though I have many unique modifications only some will be useful.

I appreciate the quick responses, I hope to be able to contribute more as I get further into the build

LoQue90, 2025-04-21:
Having someone to test the changes on a live mashine would be really helpful.
I have to look how the scale works to adapted my changes to it.

scottcondie24, 2025-04-22:
I successfully loaded a fork to github with all my changes, and saw the process of comparing to the master branch. With a bit of work I can load the changes from #512 if it doesn't interfere with my config and check that this has been addressed

LoQue90, 2025-06-02:
Fixed with #512

closed 2025-06-02 by LoQue90